STMicroelectronics/STM32CubeL4 #2: linker scripts licencing

khalifima, 2019-12-12:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus) 

**Describe the set-up**
stm32l4r5

**Additional context**
https://github.com/STMicroelectronics/STM32CubeL4/blob/master/Projects/NUCLEO-L4R5ZI/Templates/SW4STM32/STM32L4R5ZI_NUCLEO/STM32L4R5ZITx_FLASH.ld is this file really meant to be proprietary license. the other linker scripts are BSD3 
**Screenshots**
If applicable, add screenshots to help explain your problem.

ALABSTM, 2019-12-12:
Hi Khalifima,

Please let me introduce @CCASTM who will answer your question about the license aspect.

With regards,

khalifima, 2019-12-17:
@ALABSTM thanks for your reply. I consider a no answer as a yes answer. So I will write a linker script and put it as a free license EPL2. BR

closed 2019-12-17 by khalifima