STMicroelectronics/STM32CubeL4 #3: warning: 'static' is not at beginning of declaration

cracked-machine, 2020-01-02:
Hi

The definition of USBH_HID_InterfaceDeInit has the static keyword in different order to the declaration and generates a compiler warning:

../CubeMX/L476_USBHost/Middlewares/ST/STM32_USB_Host_Library/Class/HID/Src/usbh_hid.c:247:1: warning: 'static' is not at beginning of declaration [-Wold-style-declaration]
[build]   247 | USBH_StatusTypeDef static USBH_HID_InterfaceDeInit (USBH_HandleTypeDef *phost)

https://github.com/STMicroelectronics/STM32CubeL4/blob/f6877af03b5c8b666b3443f5295cf95ec9c0b15d/Middlewares/ST/STM32_USB_Host_Library/Class/HID/Src/usbh_hid.c#L247

Moving static to the start of the line fixes the warning.

ALABSTM, 2020-01-02:
Hi Cracked-Machine,

Thank you for reporting this issue. Please let me introduce @bouattay from our development teams who will take in charge the issue.

In the meanwhile, you may have noticed that clicking the "New issue" button displays a template listing different details needed to better analyze issues. Would you mind providing them please (firmware revision, IDE, compiler, and their versions). Thank you in advance for you collaboration.

With regards,

bouattay, 2020-01-03:
Hi ALABSTM,

The issue is already fixed in host lib release v3.3.3

Regards,
Aymen

CCASTM, 2020-01-08:
Hello cracked-machine,
You can find this v3.3.3 version of the USB Host in the L4 FW Package v1.15.0 published this week
Regards
Christophe

CCASTM, 2020-01-13:
Hello Ali,
I think that we can close this Issue
Regards
Christophe

ALABSTM, 2020-01-13:
Hi Cracked-Machine,

The [STM32CubeL4](https://github.com/STMicroelectronics/STM32CubeL4) **v1.15.0** firmware package has been published today including the USB Host Library v3.3.3.

We do hope this new library revision addresses the issue you pointed out and would like to thank you once more for your interest and contribution. Please allow me to close this issue now.

With regards,

closed 2020-01-13 by ALABSTM