STMicroelectronics/STM32CubeL4 #4: Stalling at the DataOutStage in usbd_core.c leads to enumeration issues under Windows 10/7

fstap, 2020-01-30:
Hi,

enumeration fails for a lot of users because of stalling of the endpoint during the usb configuration stage. The issue seems to occure with pretty much every STM32, I can confirm it for the STM32L4, STM32H7 and STM32F7 series. The effects can easily be seen by using a USB protocol analyzer like USBTrace. A quick fix is to remove the line `USBD_LL_StallEP(pdev, 0U);` completely which does not interfere with the USB functionality in any way. The folowing thread describes the bug in more detail where I also recently posted the solution.

Kind regards

fstap

ALABSTM, 2020-02-03:
Hi Fstap,

Thank you again for your reporting. I have already introduced @bouattay into the loop of the issue you opened on STM32CubeH7 repository. Hopefully you get answers soon.

With regards,

ALABSTM, 2021-03-08:
Hi @fstap,

Firmware package STM32CubeL4 **v1.17.0**, which has been published a couple of days ago, integrates **v2.7.1** of the USB Device library. The issue you pointed out should be fixed now. Actually, it has been addressed since **v2.6.0** of the library.

Thank you again for your contribution.

With regards,

closed 2021-03-08 by ALABSTM