STMicroelectronics/STM32CubeL4 #6: We have been facing some trouble using the SD card library to read and write data to and from an SD card with an STM32F429ZI

somyavatsa, 2020-05-13:
We have been facing some trouble using the SD card library to read and write data to and from an SD card with an STM32F429ZI. We are using the ff.h FatFs - Generic FAT file system module  R0.12c library. What we have observed is that after a few reads and writes, the STM goes into a hard fault crash or a FR_DISK_ERR is returned. This was observed while running the debugger. We have tried multiple ways, even mounting the SD card and unmounting it before and after every read/write attempt. We have tried with 2gb and 4gb sd cards. The SD card used was formatted using both FAT default and FAT32.

RKOUSTM, 2020-05-29:
Hi @somyavatsa,

First of all we would like to thank you for your report. However, it seems that your issue is rather related to F4 series than to L4. Could you please close this one and open the same under **STM32CubeF4** repository? Thank you.

In order to allow a better understanding of the problem, could you please give us more details : 

  1. would you, please, send us a part of your application code to transfer it to our development 
      teams for deeper analysis.
  2. could you please confirm that, when no debug task, the issue is systematically reproducible on 
      your side?

Thank you again for your contribution.

With regards,

RKOUSTM, 2020-06-08:
Hi @somyavatsa,

I hope you are doing well. Any feedback regarding this issue?

With regards,

RKOUSTM, 2020-06-17:
Hi @somyavatsa,

Any new comments to move this discussion forward?

With regards,

RKOUSTM, 2020-07-08:
Hi @somyavatsa,
 
I hope you are fine. Any new feedback regarding this issue?

With regards,

RKOUSTM, 2020-07-08:
Hi @somyavatsa,

I hope you are doing well. Although we have not received any details about your problem, this issue will be forwarded to our development teams for analysis. We will be back to you as soon as we get feedback.

Thank you for your contribution.

With regards,

RKOUSTM, 2020-07-10:
Hi @somyavatsa,

According to our development teams, without more details on this issue, we unfortunately cannot investigate this issue and speed up the analysis procedure. 

Please allow me to close this issue now. Thank you again for your report.

With regards,

closed 2020-07-10 by RKOUSTM