STMicroelectronics/STM32CubeL4 #7: CMSIS-RTOS V2 bug: incorrect implementation of osFlagsWaitAll option for osEventFlagsWait()

fronders, 2020-06-02:
**The set-up**
 * Custom board using STM32L452CCUx MCU
 * STM32CubeIDE v1.3.1 under Windows 10 x64 using STM32CubeL4 pack v1.15.1
 * FreeRTOS v10.2.1 with CMSIS-RTOS v2.00

**Describe the bug**
Incorrect implementation of `osFlagsWaitAll` option for `osEventFlagsWait()` is causing incorrect code functioning. When using `osEventFlagsWait()` function with `osFlagsWaitAll` I expect that the thread calling gets unblocked when *all* of the *specified* flags are set. However if any other flags (additional to specified) are also set - `osEventFlagsWait()` returns with `osErrorTimeout`. This happens even with `osWaitForever` timeout, which is nonsense.

Problem lies in the ST's wrapper around FreeRTOS function `xEventGroupWaitBits()`. If more flags then specified are set - the function will report error, which shouldn't happen.

Specifically this check on [cmsis_os2.c line 1179](https://github.com/STMicroelectronics/STM32CubeL4/blob/master/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2/cmsis_os2.c#L1179) is causing the bug. Here is the piece of code around those lines:
```
    rflags = xEventGroupWaitBits (hEventGroup, (EventBits_t)flags, exit_clr, wait_all, (TickType_t)timeout);
    if (options & osFlagsWaitAll) {
      if (flags != rflags) {    // <--------- THIS IS A BUG
        if (timeout > 0U) {
          rflags = (uint32_t)osErrorTimeout;
        } else {
          rflags = (uint32_t)osErrorResource;
        }
      }
    }
    else {
      if ((flags & rflags) == 0U) {
        if (timeout > 0U) {
          rflags = (uint32_t)osErrorTimeout;
        } else {
          rflags = (uint32_t)osErrorResource;
        }
      }
    }
```

**Proposed solution**
Replace the buggy condition check:
```
if (flags != rflags) {    // <--------- THIS IS A BUG
    // ...
}
```
with a correct one:
```
if ((flags & rflags) != flags) {    // <--------- THIS IS CORRECT IMPLEMENTATION
    // ...
}
```

RKOUSTM, 2020-06-03:
Hi @fronders,

Thank you for this report. It will be forwarded to our development teams for analysis. We will be back to you as soon as we get an answer. 

Thank you for your patience and thank you again for your contribution.

With regards,

rahmanih, 2020-06-03:
Hi @fronders,

this is indeed a bug! and will be fixed in the next release.

regards
Haithem.

rahmanih, 2020-06-09:
Hi @fronders 

It seems that the bug was fixed in the official [CMSIS-FreeRTOS](https://github.com/ARM-software/CMSIS-FreeRTOS) repo. (check [here](https://github.com/ARM-software/CMSIS-FreeRTOS/blob/f6270daab9b8b7a2b42ce65c1f30f545c1d1ae4b/CMSIS/RTOS2/FreeRTOS/Source/cmsis_os2.c#L811)

We are planning to update the STM32CubeFW with the latest CMSIS-FreeRTOS release to integrate the fix.

regards
Haithem.

rahmanih, 2020-06-22:
I think we can close this issue now.

regards
Haithem.

ALABSTM, 2020-09-21:
Reopened till a new STM32CubeL4 firmware version supporting FreeRTOS **v10.3.1** is published.

RKOUSTM, 2021-08-16:
Hi @fronders,

The fix you requested has been implemented and is now available in the frame of the STM32CubeL4 FW V1.17.0 package where we have upgraded to use FreeRTOS V10.3.1 version. 

This issue can be closed now. Thank you again for your contribution.

With regards,

closed 2021-08-16 by RKOUSTM