STMicroelectronics/STM32CubeL4 #8: SPI2 with LL lib  run abnormal  in STM32L431

zfz1120, 2020-07-24:
**Caution**
HARDWARE:STM32L431RC
SPI2：
//  /**SPI2 GPIO Configuration  
//  PC3   ------> SPI2_MOSI
//  PB13   ------> SPI2_SCK 
//  */

when use ll lib to init PB13 as sck，my board sometimes works abnormal;
if i chooss  hal lib to init PB13, it‘s OK.

`static void MX_SPI2_Init(void)
{

  /* USER CODE BEGIN SPI2_Init 0 */

  /* USER CODE END SPI2_Init 0 */
  LL_SPI_InitTypeDef SPI_InitStruct = {0};

  /* Peripheral clock enable */
  LL_APB1_GRP1_EnableClock(LL_APB1_GRP1_PERIPH_SPI2);
  
  LL_AHB2_GRP1_EnableClock(LL_AHB2_GRP1_PERIPH_GPIOC);
  LL_AHB2_GRP1_EnableClock(LL_AHB2_GRP1_PERIPH_GPIOB);

  LL_GPIO_InitTypeDef GPIO_InitStruct = {0};
  GPIO_InitStruct.Pin = LL_GPIO_PIN_3;
  GPIO_InitStruct.Mode = LL_GPIO_MODE_ALTERNATE;
  GPIO_InitStruct.Speed = LL_GPIO_SPEED_FREQ_VERY_HIGH;
  GPIO_InitStruct.OutputType = LL_GPIO_OUTPUT_PUSHPULL;
  GPIO_InitStruct.Pull = LL_GPIO_PULL_NO;
  GPIO_InitStruct.Alternate = LL_GPIO_AF_5;
  LL_GPIO_Init(GPIOC, &GPIO_InitStruct);

  GPIO_InitStruct.Pin = LL_GPIO_PIN_13;
  GPIO_InitStruct.Mode = LL_GPIO_MODE_ALTERNATE;
  GPIO_InitStruct.Speed = LL_GPIO_SPEED_FREQ_VERY_HIGH;
  GPIO_InitStruct.OutputType = LL_GPIO_OUTPUT_PUSHPULL;
  GPIO_InitStruct.Pull = LL_GPIO_PULL_NO;
  GPIO_InitStruct.Alternate = LL_GPIO_AF_5;
  LL_GPIO_Init(GPIOB, &GPIO_InitStruct);

//      __HAL_RCC_GPIOC_CLK_ENABLE();
//	GPIO_InitTypeDef GPIO_InitStruct1;
//    GPIO_InitStruct1.Pin = GPIO_PIN_13;
//    GPIO_InitStruct1.Mode = GPIO_MODE_AF_PP;
//    GPIO_InitStruct1.Pull = GPIO_NOPULL;
//    GPIO_InitStruct1.Speed = GPIO_SPEED_FREQ_VERY_HIGH;
//    GPIO_InitStruct1.Alternate = GPIO_AF5_SPI2;
//    HAL_GPIO_Init(GPIOB, &GPIO_InitStruct1);

  SPI_InitStruct.TransferDirection = LL_SPI_FULL_DUPLEX;//LL_SPI_HALF_DUPLEX_TX;//
  SPI_InitStruct.Mode = LL_SPI_MODE_MASTER;
  SPI_InitStruct.DataWidth = LL_SPI_DATAWIDTH_8BIT;
  SPI_InitStruct.ClockPolarity = LL_SPI_POLARITY_HIGH;
  SPI_InitStruct.ClockPhase = LL_SPI_PHASE_2EDGE;
  SPI_InitStruct.NSS = LL_SPI_NSS_SOFT;
  SPI_InitStruct.BaudRate = LL_SPI_BAUDRATEPRESCALER_DIV2;
  SPI_InitStruct.BitOrder = LL_SPI_MSB_FIRST;
  SPI_InitStruct.CRCCalculation = LL_SPI_CRCCALCULATION_DISABLE;
  SPI_InitStruct.CRCPoly = 7;  
	if (LL_SPI_Init(SPI2, &SPI_InitStruct) != SUCCESS)
	{
		HardFault_Handler();
	}
 
  LL_SPI_SetStandard(SPI2, LL_SPI_PROTOCOL_MOTOROLA);
  LL_SPI_EnableNSSPulseMgt(SPI2);
	LL_SPI_Enable(SPI2);

}
`
**Describe the set-up**
 * The board (either ST RPN reference or your custom board).
 * IDE or at least the compiler and its `version.`

**Describe the bug**
A clear and concise description of what the bug is.

**How To Reproduce**
1. Indicate the global behavior of your application project.

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).

3. The use case that generates the problem.

4. How we can reproduce the problem.

**Additional context**
If you have a first analysis or patch correction, thank you to share your proposal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

ASELSTM, 2020-07-28:
Hi @zfz1120,

Thank you for this report. In order to get a better understand of your issue, we need more clarifications : 

1. Could you please provide us with more details about the abnormal behavior observed on your board.
 
2.  Would you try to change the SPI2_SCK pin configuration as following and check whether you still have the same behavior : 

```
  LL_GPIO_SetPinMode(GPIOB, LL_GPIO_PIN_13, LL_GPIO_MODE_ALTERNATE);
  LL_GPIO_SetAFPin_8_15(GPIOB, LL_GPIO_PIN_13, LL_GPIO_AF_5);
  LL_GPIO_SetPinSpeed(GPIOB, LL_GPIO_PIN_13, LL_GPIO_SPEED_FREQ_HIGH);
  LL_GPIO_SetPinPull(GPIOB, LL_GPIO_PIN_13, LL_GPIO_PULL_DOWN);
```
With regards,

ASELSTM, 2020-08-11:
Hi @zfz1120,

Any feedback?

Thank you,

ASELSTM, 2020-09-08:
Hi @zfz1120,

Thank you once more for your contribution. However, without more details we won't be able to investigate this issue further.
So, please allow me to close this thread as no activity has been registered since a month. You can reopen it at any time if you have details to provide us to help you solve this issue.
Thank you for your comprehension.

With regards,

closed 2020-09-08 by ASELSTM