STMicroelectronics/STM32CubeL4 #9: Clone not possible ?

m0nka, 2020-08-04:
Hello,

It seems there is an issue with cloning the repo. I have tried with:

- STM32CubeIDE
- SourceTree
- GitHub desktop

Here the error message from last tool:

Cloning into 'C:\STM32CubeL4'...
remote: Enumerating objects: 37019, done.        
error: RPC failed; curl 56 Failure when receiving data from the peer            
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed

All other tools give this "early EOF" error.

Thank you!

K.

ALABSTM, 2020-08-04:
Hi @m0nka,

Does the issue still persist? It may happen sometimes when not enough bandwidth is left for the connection to the servers. The simplest way to deal with this is to wait for a while and retry the operation till it succeeds. I hope this helps.

With regards,

m0nka, 2020-08-04:
Hey,

Thanks for the fast response. Just tried. Still getting the same error. Will try tomorrow and will let you know.

BR

m0nka, 2020-08-06:
Ok, works now. Thank you!

closed 2020-08-06 by m0nka