STMicroelectronics/STM32CubeL4 #10: Using Microphone with B-L475E-IOT01A

sundar19, 2020-08-19:
I tried to find examples for using microphone in the discovery iot node , but couldnt find them. Is there any example codes for using mic in that dev board ?
Thanks!

RKOUSTM, 2020-08-25:
Hi @sundar19,

Thank you for your contribution.  We don't have an audio example  for B-L475E-IOT01A  board but you can adapt [this](https://github.com/STMicroelectronics/STM32CubeL4/tree/master/Projects/32L476GDISCOVERY/Examples/DFSDM/DFSDM_AudioRecord) DFSDM audio example to your code.

Thank you again for your contribution.

 With regards,

closed 2020-09-08 by RKOUSTM

CIPop, 2021-12-16:
+1 - I found that audio drivers for this board are available here: https://github.com/STMicroelectronics/x-cube-iota1/blob/main/Drivers/BSP/B-L475E-IOT01/stm32l475e_iot01_audio.h