STMicroelectronics/STM32CubeL4 #12: B-L475E-IOT01A proximity code not working

sundar19, 2020-08-24:
**Describe the set-up**
 * B-L475E-IOT01A
 *STM32cube IDE

**Describe the bug**
The proximity example builds without any error ,after uploading the code into the board and opening a serial monitor to press "n" button to view the proximity value after which the code suddenly breaks and stops.

[image]

I suspect some error in VL53L0X_PROXIMITY_GetDistance() , how to rectify this error
Thanks!!!

RKOUSTM, 2020-08-26:
Hi @sundar19,

Welcome again to our GitHub organization. The problem seems to be related to migration to the CubeIDE. However, this public organization on GitHub is intended to promote co-development and issues posted here shall exclusively be related to the software published within the different repositories.
For any other issue (e.g. related to documentation, to hardware, to ecosystem, to user application, etc.), you shall address your questions to the [ST Community](https://community.st.com/s/topiccatalog). There you shall look for the adequate topic and submit your question or remark.

You can use the IDE proposed by ST (IAR, MDK-ARM and SW4STM32) for Proximity application on B-L475E-IOT01A all of them are working well.

Thank you for you contribution.

With regards,

closed 2020-08-26 by RKOUSTM

sundar19, 2020-08-26:
Thank you so much!