STMicroelectronics/STM32CubeL4 #13: STM32L433 quartzless HSI RC 16 default calibration problem

christophemonmon, 2020-08-27:
I use an **STM32L433** and I upgraded from version 1.14.0 to the 1.16.0 package. 

My IDE is Keil (IDE-Version : µVision V5.31.0.0)

There is no quartz and I have a USB and CAN connection. So I use the HSI RC 16MHz to generate SYSCLK. For USB, I use the 48MHz RC.

I encountered a problem on the CAN and it turns out that the default calibration has been changed in stm32l4xx_hal_rcc.h.
By default, it should be 16 and it's set to 64 so the bits generated on my 1MHz CAN link was about 0.96ns instead of 1ns. Resynchronization was therefore impossible.

Line 208 :
#if defined(RCC_ICSCR_HSITRIM_6)
#define RCC_HSICALIBRATION_DEFAULT 0x40U /*!< Default HSI calibration trimming value 64 on devices other than STM32L47x/STM32L48x */
#else
#define RCC_HSICALIBRATION_DEFAULT 0x10U /*!< Default HSI calibration trimming value 16 on STM32L47x/STM32L48x devices */
#endif /* RCC_ICSCR_HSITRIM_6 */

To correct temporarily, I changed the 0x40 to 0x10.

RKOUSTM, 2020-08-31:
Hi @christophemonmon,

Thank you are contribution. It will be forwarded to our development teams for analysis.I will get back to you as soon as they provide me with their feedback.

With regards,

RKOUSTM, 2020-10-08:
Hi @christophemonmon,

Thank you for having reported this point. This issue is confirmed by our development teams. A fix will be implemented and published in the frame of a future release. Thank you once more. 

With regards,

RKOUSTM, 2020-10-08:
ST Internal Reference: 93291

adentse, 2020-11-30:
Thank you. I'am very glad I found this thread or I would have torn my hair in despair. My UART was one bit off after updating the IDE.

boyhuesd, 2021-01-15:
Hi, did it get fixed? If yes, may I know where is the diff? Thank you.

RKOUSTM, 2021-03-04:
Hi @boyhuesd,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.17.0 of the STM32CubeL4 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-04 by RKOUSTM