STMicroelectronics/STM32CubeL4 #14: STM32L433 HID USB problem

christophemonmon, 2020-08-27:
I use an STM32L433 and I upgraded from version 1.14.0 to the 1.16.0 package. 

There is no quartz and I have a USB and CAN connection. So I use the HSI RC 16MHz to generate SYSCLK. For USB, I use the 48MHz RC.

I encountered a problem with HID on the USB.

In the end, I found that a line is missing in the USBD_CUSTOM_HID_DataOut function of the usbd_customhid.c file.

To correct the problem I put the following line before the return :
```
  USBD_LL_PrepareReceive(pdev, CUSTOM_HID_EPOUT_ADDR , hhid->Report_buf,
                         USBD_CUSTOMHID_OUTREPORT_BUF_SIZE);

  return (uint8_t)USBD_OK;
}
```

ALABSTM, 2020-09-16:
ST Internal Reference: 45118

ALABSTM, 2020-09-16:
Hi @christophemonmon,

Thank you for this report and for the solution suggested. Actually,  the call to `USBD_LL_PrepareReceive()` is now made separately in the new function `USBD_CUSTOM_HID_ReceivePacket()`.

In you application, simply call `USBD_CUSTOM_HID_ReceivePacket()` just after `USBD_CUSTOM_HID_DataOut()`.

With regards,

ALABSTM, 2020-09-21:
Hi @christophemonmon,

Does the solution above addresses your need? Thank you for you feedback.

With regards,

ALABSTM, 2020-10-12:
Hi @christophemonmon,

Any feedback? Is your issue solved with the suggested solution?

With regards,

ALABSTM, 2020-10-13:
Hi @christophemonmon,

I think this issue can be closed now. Indeed, this new update of the `usbd_customhid.c` source file requires you to update your application. You can reopen it if you have anything else to add regarding this point.

With regards,

closed 2020-10-13 by ALABSTM