STMicroelectronics/STM32CubeL4 #15: Pointer function parameters should be const if the content is not modified

MarioMalenica, 2020-10-27:
There is an issue with API design, and its really visible with drivers.

If a function parameter is a pointer, and if a function is not modifying the resource that the pointer is pointing to, then that pointer should be declared as "const

Users of your API need to jump through hoops, or insert ugly #pragmas if they need to compile the code with strict settings and with all + extra warnings enabled.

A good example for this is HAL_I2C_Master_Transmit().
Why "uint8_t *pData" is not "const uint8_t *pData"?

With Keil 5 and ARM compiler 6 and in a relatively simple and mid-sized project, if I enable all compiler warnings I'm seeing in average about 120 compilation warnings for latest version (1.16) of this library.

ALABSTM, 2020-11-17:
Duplicate of [STM32CubeF4#10](https://github.com/STMicroelectronics/STM32CubeF4/issues/10).

ALABSTM, 2020-11-17:
Hi @MarioMalenica,

Thank you for your report. We have planned a global update of the HAL and LL drivers since a while. However, this is likely to require some time to be deployed. We cannot share a date for the moment, but we will do our best to have such a global update deployed and released soon. We count on your patience.

Thank you again for having tackled the subject.

With regards,

ALABSTM, 2023-12-12:
ST Internal Reference: 137863

TOUNSTM, 2024-07-29:
Fixed in 82e7ade76252f4c973c9fc7515ebc42b850b7c0b

closed 2024-07-29 by TOUNSTM