STMicroelectronics/STM32CubeL4 #18: const as much as possible

atsju, 2020-11-10:
https://github.com/STMicroelectronics/STM32CubeL4/blob/d023c0d560ace11509f9b761c8913a9e48fcf194/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_hal_crc.c#L339

Should be : uint32_t HAL_CRC_Calculate(CRC_HandleTypeDef *hcrc, **const** uint32_t pBuffer[], uint32_t BufferLength) 
This tells the user and the compiler that function will not modify the content of buffer. There are other occurrence in the HAL. Please check them all.

Note this is not the same as #17.

RKOUSTM, 2020-11-12:
Hi @atsju,

Thank you for this detailed report. This issue has also been previously reported by other users ([link](https://github.com/STMicroelectronics/STM32CubeF4/issues/10)).

This is one of the main points in the tasks queue of our development teams. We cannot share a date regarding its deployment and publication. We count on your patience and comprehension.

Thank you again for this report. 

With regards,

RKOUSTM, 2020-11-19:
Hi @atsju,
 
First, allow me to thank you for your contribution. The final decision of our technical committee is no plan to add a const inside the function. The reported point is already highlighted during the MISRA-C 2012 compliance rework (Rule-8.13). It has been decided, from the quality point of view, to derogate this rule as fixing it will break the compatibility with almost all HAL APIs and then update all examples and applications within the Cube deliverable and also on user application side.

Please, allow me to close this issue. Thank you for your comprehension.

With regards,

closed 2020-11-19 by RKOUSTM

atsju, 2020-11-19:
@RKOUSTM I answered there : https://github.com/STMicroelectronics/STM32CubeF4/issues/10

ST is quite new to open-sourcing the code on Github and you are doing well.
I just wanted to give you a kind advice (take it or not): note that other projets close issues as soon as they are spotted to be duplicate. This permits to centralise discutions to original ticket and to see the importance of the issue. It also reduces your maintenance task when a decision is made about a ticket it is only one and not all your duplicates.

ALABSTM, 2023-08-07:
Duplicate of #17