STMicroelectronics/STM32CubeL4 #19: cast pointer to a bigger data is danger

CanastraRF, 2020-11-22:
Dear Support
In different functions you cast a uint8_t* to uint16_t* or uint32_t*. This can result in a hardfault.

Examples from stm32l4xx_hal_spi.c:

```
HAL_StatusTypeDef HAL_SPI_Transmit(SPI_HandleTypeDef *hspi, uint8_t *pData, uint16_t Size, uint32_t Timeout)
{
…
      hspi->Instance->DR = *((uint16_t *)hspi->pTxBuffPtr);
…

```
pTxBuffPtr can point to an odd or even address. With a odd address a hardfault is generated.
So please handle this correct.

ASELSTM, 2020-12-15:
Hi @CanastraRF,

Thank you for this report. Your request has been forwarded to our development teams. We will get back to you as soon as they provide us with their feedback.

With regards,

ASELSTM, 2021-07-08:
Hi  @CanastraRF,

Excuse this meanwhile delayed response. Our development team has been working to find a workaround to the issue you pointed out but unfortunately all the possible solutions are leading either to a drop of the performance or to an increase of the memory consumption which can't be accepted.

However, our development team is suggesting you to provide an aligned data to the `hspi-> Init.DataSize` by setting it to SPI_DATASIZE_8BIT or SPI_DATASIZE_16BIT. They are also proposing the use of  `ALIGN16` macros at application level.

Please allow me then to close this thread. If you have any updates you can reopen it at any time. Thank you for your comprehension.

With regards,

closed 2025-12-23 by ALABSTM