STMicroelectronics/STM32CubeL4 #20: Question about emmc speed in stm32l4p5g-dk

fito-jaeuklee, 2020-11-24:
Hello! 

I'm bought new DK board that name STM32L4P5G-DK for testing Fatfs-eMMC example.
( https://github.com/STMicroelectronics/STM32CubeL4/tree/master/Projects/32L4P5GDISCOVERY/Applications/FatFs/FatFs_eMMC_Standalone )

It works fine, but not the speed I expected. 
Almost datasheet saying that eMMC speed is more than 50MB/s. But real speed is 1MB/s in this example .... :(

I would like to know if there are other ways to improve the speed.

i check up this write/read time

`copy_start = (uint32_t *)0x20002000;
    /* Create and Open test files object with write access to measure eMMC read/write accesses */
    for (counter = 0; counter< N_SIZES; counter++)
    {
      if(f_open(&MyFile, "TEST_ACCESS_TIME.TXT", FA_CREATE_ALWAYS | FA_WRITE) == FR_OK)
      {

        /* Start timer counter */
        HAL_TIM_Base_Start(&htim);
        /* Write data to the text file */
         res = f_write(&MyFile, copy_start, data_sizes[counter], (void *)&byteswritten);
        /* Stop timer counter */
        HAL_TIM_Base_Stop(&htim);
        Access_time_us = __HAL_TIM_GET_COUNTER(&htim);
        print_spend_time_clock_write = Access_time_us /= SystemClock;
        read_bytes = byteswritten;
        __HAL_TIM_SET_COUNTER(&htim, 0x0);
//        printf("%ld\t%s\t%ld\t%s\n", Access_time_us,"us to write",byteswritten,"bytes");

        if((byteswritten > 0) && (res == FR_OK))
        {
          /* Close the open text file */
          f_close(&MyFile);

          /* Open the text file object with read access */
          if(f_open(&MyFile, "TEST_ACCESS_TIME.TXT", FA_READ) == FR_OK)
          {
            /* Start timer counter */
            HAL_TIM_Base_Start(&htim);
            /* Read data from the text file */
            res = f_read(&MyFile, copy_start, data_sizes[counter], (void *)&bytesread);
            /* Stop timer counter */
            HAL_TIM_Base_Stop(&htim);
            Access_time_us = __HAL_TIM_GET_COUNTER(&htim);
            print_spend_time_clock_read = Access_time_us /= SystemClock;
            read_bytes = byteswritten;
            __HAL_TIM_SET_COUNTER(&htim, 0x0);
//            printf("%ld\t%s\t%ld\t%s\n\n", Access_time_us,"us to read",byteswritten,"bytes");

            if((bytesread > 0) && (res == FR_OK))
            {
              /* Close the open text file */
              f_close(&MyFile);

              /* Compare read data with the expected data */
              if((bytesread != byteswritten))
              {
                Error_Handler();
              }
            }
          }
        }
      }`

fito-jaeuklee, 2020-11-24:
This capture is Stm32l4p5g-dk built-in chip emmc data speed from datasheet.
[image]

RKOUSTM, 2020-12-11:
Hi @fito-jaeuklee,

Thank you for your contribution.  To improve the eMMC speed, you can call [BSP_MMC_ConfigSpeedBusOperation ()](https://github.com/STMicroelectronics/STM32CubeL4/blob/d023c0d560ace11509f9b761c8913a9e48fcf194/Drivers/BSP/STM32L4P5G-Discovery/stm32l4p5g_discovery_mmc.c#L166-L187) in your application just inside function [MMC_initialize()](https://github.com/STMicroelectronics/STM32CubeL4/blob/d023c0d560ace11509f9b761c8913a9e48fcf194/Projects/32L4P5GDISCOVERY/Applications/FatFs/FatFs_eMMC_Standalone/Src/mmc_diskio.c#L73-L90function) as follows:

```diff

DSTATUS MMC_initialize(BYTE lun)
{

    if (MMC_OK == BSP_MMC_DeInit())
    {
      if (MMC_OK == BSP_MMC_Init())
      {
+       BSP_MMC_ConfigSpeedBusOperation(SDMMC_SPEED_MODE_HIGH);
        Stat = MMC_CheckStatus(lun);
      }
    }

   return Stat;
}
```
Please let us know your feedback so that we can help you.

With regards,

closed 2020-12-16 by RKOUSTM