STMicroelectronics/STM32CubeL4 #21: Add .IOC file

ferraridamiano, 2020-11-25:
These example are very helpful, but all the .ioc files are missing. Is it possibile to have them?

Thank you!

ASELSTM, 2020-12-16:
Hi @ferraridamiano,

First we would like to thank you for your contribution. 

Actually, for legacy series, the IOC file is not delivered within the firmware package. We understand that this is binding and that they would be helpful, but the IOC files are only delivered in the firmware package of recent series. Therefore, our development team regrets to decline your request. Thank you for your understanding.

Please allow me then to close this thread. Thank you for your comprehension and thank you again for your contribution. Hope to hear from you again.

With regards,

closed 2020-12-16 by ASELSTM

jeromecoutant, 2020-12-16:
Hi
Maybe you could check
https://github.com/STMicroelectronics/STM32_open_pin_data/tree/master/boards