STMicroelectronics/STM32CubeL4 #23: STM32L431 with X-CUBE-EEPROM_V3.0.0. Problem writing to flash memory.

christophemonmon, 2020-12-03:
I use **Keil v5.33** and **STM32Cube_FW_L4_V1.16.0**

The project uses a **STM32L431** with X-CUBE-EEPROM.

First, the X-CUBE-EEPROM extension cannot be integrated automatically in STM32CubeMX (v6.1.0). The application answers that the zip is not compliant. So you have to manually add the files you need in the project. It's not complicated but a bit long to identify the few files you need in all directories.

Second problem already encountered with the **STM32F405** for a **booloader** function: before you can write in flash, you have to delete several configuration bits of the flash even if they are already at 0 (seen in debug mode with keil). 

Addition of :
`__HAL_FLASH_CLEAR_FLAG(FLASH_FLAG_EOP | FLASH_FLAG_OPERR | FLASH_FLAG_WRPERR | FLASH_FLAG_PGAERR | FLASH_FLAG_PGPERR|FLASH_FLAG_PGSERR); `
in the **ProgramFlash** function on line function 103

For the **STM32L431** with EEPROM emulation, I added this line in the `eeprom_emul.c` function:
`__HAL_FLASH_CLEAR_FLAG(FLASH_FLAG_EOP | FLASH_FLAG_OPERR | FLASH_FLAG_WRPERR | FLASH_FLAG_PGAERR | FLASH_FLAG_PGSERR);`
before the FI_WriteDoubleWord on line 1560.

Hoping to help people.

RKOUSTM, 2020-12-11:
Hi @christophemonmon,

Thank you for contribution. It has been transferred to our technical teams for deeper analysis. We will be back to you as soon as we get an answer. Thank you for you again for you contribution.

With regards

RKOUSTM, 2021-01-06:
Hi @christophemonmon,

Thank you for your report. For the first point, our development team are started to work on making `X-CUBE-EEPROM` package compatible with`CubeMX`. We cannot share a date for the moment, but we will do our best to have such a global update deployed and released soon. Concerning the second point, the `HAL_FLASH_Program` function already clears all the error flags if at least one has been raised, before each operation (as for the L4 HAL it is done in `FLASH_WaitForLastOperation -> FLASH_SetErrorCode ->__HAL_FLASH_CLEAR_FLAG(FLASH_FLAG_ALL_ERRORS)`). This corresponds to the flash programming procedure exposed in the reference manual. In order to allow a better analysis of this point, could you please give us more details about the problem you encountered without resetting the error flags again, so that our development teams can give you an answer.

Thank you for your contribution. 

with regards,

RKOUSTM, 2021-02-09:
Hi @christophemonmon,

I hope you are doing well. Any new comments to move this discussion forward?

With regards,

With regards,

christophemonmon, 2021-02-09:
Hi @RKOUSTM ,

I'm sorry for the late response, the messages were going as spam.
To answer your question, if I don't clear the flags, the program goes into hard fault, it's crashed!
If I look at the flags with the debugger (with keil ulink pro probe), they are already at 0! So I don't understand this behavior.

Best regards.

RKOUSTM, 2021-02-10:
Hi @christophemonmon,

Thank you for your feedback. Please allow me to introduce @ASASTM who will look deeper into the subject.

With regards,

ASASTM, 2021-02-11:
Hi @christophemonmon,

What does your program going in hardfault exactly ? 
When you run the STM32Cube_FW_L4_V1.16.0\Projects\NUCLEO-L412RB-P\Examples\FLASH\FLASH_EraseProgram\MDK-ARM project, the flash programming operations are done without problem.

Best regards,
Adam

RKOUSTM, 2021-04-02:
Hi @christophemonmon,

I hope you are fine. Any new feedback regarding the request of my colleague @ASASTM ? 

With regards,

RKOUSTM, 2021-04-26:
Hi @christophemonmon,

Please allow me to close this thread. You can reopen it at any time if you have details to provide us to help you.

Thank you for your comprehension and thank you once more for your contribution

With regards,

closed 2021-04-26 by RKOUSTM