STMicroelectronics/STM32CubeL4 #24: STM32L4 LPTIM as pulse counter of intermittent pulses fails when initialising with no pulse

trunet, 2020-12-24:
**Describe the set-up**
 * NUCLEO-L432KC
 * STM32CubeIDE 1.5.0
 * [AN4865](https://www.st.com/resource/en/application_note/dm00290631-lowpower-timer-lptim-applicative-use-cases-on-stm32-microcontrollers-stmicroelectronics.pdf)
 * Anemometer that generates pulses when it's winding

**Describe the bug**
When `HAL_LPTIM_Counter_Start(&hlptim1, PERIOD);`, it HAL_TIMEOUT, unless it's windy or I turn the anemometer with my hand. Note that if it timeout, the read counter is always zero. When it successfully start, even if I have zero pulses in a long time, when there is, it works fine.

**How To Reproduce**
1. Follow [AN4865](https://www.st.com/resource/en/application_note/dm00290631-lowpower-timer-lptim-applicative-use-cases-on-stm32-microcontrollers-stmicroelectronics.pdf) and instead of hooking up on a signal generator with stable pulses, connect to intermittent pulses like an anemometer or a rain gauge. All the rest is the same, like LSE clock driving LPTIM.

2. It will not start the timer with HAL_TIMEOUT, if there's no pulse. `HAL_LPTIM_ReadCounter()` returns 0.

**Additional context**
This is probably related to [stm32l4xx_hal_lptim.c](https://github.com/STMicroelectronics/STM32CubeL4/blob/master/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_hal_lptim.c#L1638) HAL_LPTIM_Counter_Start function on this code:

```
  /* Load the period value in the autoreload register */
  __HAL_LPTIM_AUTORELOAD_SET(hlptim, Period);

  /* Wait for the completion of the write operation to the LPTIM_ARR register */
  if (LPTIM_WaitForFlag(hlptim, LPTIM_FLAG_ARROK) == HAL_TIMEOUT)
  {
    return HAL_TIMEOUT;
  }
```

If you comment the if block, whenever you have >5 pulses (as the datasheet says) `HAL_LPTIM_ReadCounter()` starts incrementing the pulses.

The AN and the datasheet, where it describes LPTIM as external pulse counter, doesn't say anything about LPTIM_ARR register neither LPTIM_ISR waiting for ARROK flag (4th bit).

ASELSTM, 2021-01-13:
Hi @trunet,

Thank you for this detailed report. 

Referring to the reference manual, LPTIM can be clocked by an external clock through the LPTIM external input1 or clocked by an internal clock source (APB clock or any of the embedded oscillators). In the first use case, it is necessary to ensure a stable signal for the LPTIM to work properly. 

In the other hand,  the check on the `LPTIM_FLAG_ARROK` is necessary in order to make sure that the period value is well set in the  LPTIM_ARR register.  Thus, to ensure a correct behavior of your system this check is important so that it is a guarantee that the LPTIM is well clocked and that it is functional.

In order to solve your issue, I suggest you to clock your LPTIM on an internal clock  (`LptimHandle.Init.Clock.Source = LPTIM_CLOCKSOURCE_APBCLOCK_LPOSC`) and configure the counter to increment at each external event coming from your anemometer (`LptimHandle.Init.CounterSource   = LPTIM_COUNTERSOURCE_EXTERNAL`). 

Hope this would help you to solve the issue.

With regards,

ASELSTM, 2021-02-02:
Hi @trunet,

Please allow me to close this thread as no activity has been registered since a month. You can reopen it at any time if you have details to provide us to help you solve this issue.
Thank you for your comprehension and thank you once more for your contribution

With regards,

closed 2021-02-02 by ASELSTM

tcpipchip, 2021-10-05:
Hi, i am having the same problem with STM32L071, using STM32 CUBE and KEIL :(

tcpipchip, 2021-10-05:
I had to add 5 to the initial count

JLH64-dev, 2023-08-05:
Was there ever a resolution to this issue? I'm trying to put together a very similar project, except with a turbine flow-meter at the front end -  no flow, no-go.
Same problem here - LpTIM won't start unless there's an external clock up and running. Chocolate teapot in real-world application.
We can comment out the error handler in the ARR routine, but it's messy. I've tried various other solutions, to no avail. Next step will be to start the LpTIM in internal clock mode, then see if we can change the input pin once it's started up.

tcpipchip, 2023-08-05:
Please, send a e-mail and i will send you my KEIL project (STM32L072)
I used to pulse water measure!
I am Miguel

tcpipchip, 2023-08-05:
tcpipchip@hotmail.com

JLH64-dev, 2023-08-05:
Hi Miguel,

Thanks for your reply.

A copy of your project would be great for reference. We are using STM_IDE, giving at a go anyway at the start of this new project. Our last big project was done under IAR.

I’m the project manager / hardware designer / team-lead on this one. I’m getting the hardware platform prepared ready before I hand the platform over for serious software dev to begin. LpTIM is the way to go, once we can get it to go.

Thanks again.

John

www.Jlhsystems.co.uk <http://www.Jlhsystems.co.uk> 

*	My website probably needs certs updating – no threats if you want to look.
*	Regards,
*	John

tcpipchip, 2023-08-05:
HI!!!!!!!!!
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/zip_16x16.png]TXCO.zip<https://1drv.ms/u/s!Ak_-1v7Ug-k2gZQyB7ZJWiyJGONPew?e=aFYwH2>

tcpipchip, 2023-08-05:
CONFIRM ME IF YOU RECEIVED

JLH64-dev, 2023-08-05:
Hi Miguel.

Yes, I received it thanks and have had a quick look in main.

There’s a big wedding coming up. I’ll get back to you when I have time for a better study.

Thanks again,

John