STMicroelectronics/STM32CubeL4 #25: Typo in comments of applications

atsju, 2021-01-02:
https://github.com/STMicroelectronics/STM32CubeL4/blob/d023c0d560ace11509f9b761c8913a9e48fcf194/Projects/32L4P5GDISCOVERY/Applications/USB_Device/MSC_Standalone/Src/main.c#L144
should be` /* Enable the HSI48 Oscillator */ `

https://github.com/STMicroelectronics/STM32CubeL4/blob/d023c0d560ace11509f9b761c8913a9e48fcf194/Projects/32L4P5GDISCOVERY/Applications/USB_Device/MSC_Standalone/Src/main.c#L172
should be ` /* Select HSI48 output as USB clock source */ `

A quick search to HSI84 finds 24 results so please have a look also to other files using the same initialisation code.

ALABSTM, 2021-01-19:
Hi @atsju,

Thank you for this report. **23** files have been identified to be containing the *HSI84* typo so far. The issue has been logged in our internal database. A fix will be made available in a future release.

With regards,

ALABSTM, 2021-01-19:
ST Internal Reference: 99723

atsju, 2021-01-19:
@ALABSTM I should have done 2 tickets. 
Please confirm that you have taken into account LSE/HSI48 mixup and not only the HSI84 typo. (there are 2 references in my ticket)

ALABSTM, 2021-01-20:
Hi @atsju,

Yes I reported both typos: LSE and HSI84. I also reported that among the 23 files identified, 6 belong rather to demos than to applications. Thank you again for your contribution and involvement.

With regards,

ALABSTM, 2022-01-12:
Hi @atsju,

This issue has been fixed in the frame of version **v1.17.0** of the STM32CubeL4 published some time ago. Please allow me to close it.

With regards,