STMicroelectronics/STM32CubeL4 #26: CMSIS headers define LTDC_Layer2, but LTDC_Layer2_BASE doesn't exist

benedekkupper, 2021-01-16:
**Describe the set-up**
 * Include stm32l4p5xx.h header
 * Try to use LTDC_Layer2 definition

**Describe the bug**
Compilation fails since there's no definition for LTDC_Layer2_BASE.

RKOUSTM, 2021-01-19:
Hi @benedekkupper,

Thank you for this report. According to reference manual [RM0432](https://www.st.com/resource/en/reference_manual/dm00310109-stm32l4-series-advanced-armbased-32bit-mcus-stmicroelectronics.pdf), we have only one layer for STM32L4P5xx/ STM32L4Q5xx and two layers for STM32L4R7xx/STM32L4R9xx and STM32L4S7xx/STM32L4S9xx. 

The reported point has been forwarded to our development teams and see if there is an update in the ``stm32l4p5xx.h`` and ``stm32l4q5xx.h`` files to correspond the reference manual.  We will get back to you as soon as they provide us with their feedback.

Will keep you informed.

With regards,

RKOUSTM, 2021-01-19:
Hi @benedekkupper,

Thank you for this report. The issue has been confirmed by our development teams. The ``LTDC_Layer2`` definition will be removed from the ``stm32l4p5xx.h`` and ``stm32l4q5xx.h`` CMSIS files. This fix will be made available in a future version. Thank you again.

Thank you again for your contribution.

With regards,

RKOUSTM, 2021-01-19:
ST Internal Reference: 99731

RKOUSTM, 2021-03-04:
Hi @benedekkupper,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.17.0 of the STM32CubeL4 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-04 by RKOUSTM