STMicroelectronics/STM32CubeL4 #28: SDMMC with dma ( with or withour freertos ) don't work

helmi-haddad-egm, 2021-02-02:
after configuration of Sdmmc prephiral with stm32cubeMx for nucelo board L452RE and a costum board with an mcu stm32L462ve , the driver generated don't work .
even when i use a project compiled for anthoer STM31L4 board and used with the other board don't work .
when i descativate the DMA for SDmmc , the driver work correctly and i can use my sd card

ASELSTM, 2021-02-04:
Hi @Hhdd06,

Thank you for this report.  For a better understanding of the problem may I ask you some question : 

1. When you say   "_the driver generated don't work_"  does this mean that there is compilation errors or the behavior of the SDMMC is not what is expected. 

2.  If possible, would you please share the generated project for further analysis.

With regards,

Hhdd06, 2021-02-09:
i mean the behavior of SDMMC when i activate the dma with CubeMX : when i use fopen function . it return FR_DISK_ERR always
but when i deactivate the dma for SDMMC , the application ( write and read in a file) work correctly .
note : even when i activate freertos which forced dma, i see the same problem .
[Sb_uSD.zip](https://github.com/STMicroelectronics/STM32CubeL4/files/5952516/Sb_uSD.zip)

ASELSTM, 2021-03-02:
Hi @Hhdd06,

After checking out the sent project, it seems like CubeMX is not importing the right files according to the configuration of FATFS. Indeed, when enabling _"Use dma template"_ within the FATAFS _"Advanced settings"_ , CubeMX should include **sd_diskio_dma.h/c**  files instead of including **sd_diskio.h/c** files so that you can use SDMMC in dma mode.

You can  check the FATFS application [FatFs_uSD_DMA_Standalone](https://github.com/STMicroelectronics/STM32CubeL4/tree/master/Projects/STM32L4R9I-EVAL/Applications/FatFs/FatFs_uSD_DMA_Standalone) proposed within the repository. It's working well and there is no issue with SDMMC in dma mode. This issue is therefore related to **CubeMX** rather than to the firmware published within this repository.

Actually, aspects related to CubeMX tool are not handled at GitHub level. They are rather treated at **ST Community** level, precisely within the CubeMX dedicated [section](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx). There, you may find posts that already raised the issue you are pointing out here. Otherwise, you can create a new post to submit your issue.

As this issue is not directly related to software component published within this repository (CMSIS, HAL, BSP, etc.) but rather to ecosystem (CubeMX tool), please allow me to close it. Thank you for your comprehension.

With regards,

closed 2021-03-02 by ASELSTM

smartel99, 2021-04-29:
After having this bug as well, I have narrowed the issue down to  the `sd_diskio` files created by the cube when using FATFS with SDMMC. I don't know exactly what the bug is since I do not have the time to troubleshoot a third-party library, so my solution was to simply get rid of the SDMMC layer and use the SPI instead. Since this is not an option offered by CubeMX, you need to select "User Defined" in FATFS, then use [this](https://github.com/kiwih/cubeide-sd-card) library.

Looking around online for solutions, I've noticed people complaining about this exact issue since at least 2016. Kind of disappointing to see that the issue has not been solved *5 years later*