STMicroelectronics/STM32CubeL4 #29: WWDG comments reference APIs that do not exist

DesktopMan, 2021-02-05:
HAL_WWDG_Start() and HAL_WWDG_Start_IT() do not exist yet are referenced in the comments. Comments should be updated to reflect correct WWDG usage. It looks like these were removed in the V1.8.0 / 25-November-2016 release.

https://github.com/STMicroelectronics/STM32CubeL4/blob/d023c0d560ace11509f9b761c8913a9e48fcf194/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_hal_wwdg.c#L51

https://github.com/STMicroelectronics/STM32CubeL4/blob/d023c0d560ace11509f9b761c8913a9e48fcf194/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_hal_wwdg.c#L56

ASELSTM, 2021-02-05:
Hi @DesktopMan,

Thank you for this report.  Actually, the point you raised has already been dealt and fixed internally. The fix will be made available in the frame of a future release.

We cannot share a date for the moment.  So stay tuned and thank you once more for your contribution and for your patience.

With regards

ASELSTM, 2021-02-05:
ST Internal Reference: 82774

ASELSTM, 2021-03-04:
Hi @DesktopMan,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.17.0 of the STM32CubeL4 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-04 by ASELSTM