STMicroelectronics/STM32CubeL4 #31: LL ADC has duplicated arguments

vpetrigo, 2021-02-17:
Hello,

[This part](https://github.com/STMicroelectronics/STM32CubeL4/blob/master/Drivers/STM32L4xx_HAL_Driver/Inc/stm32l4xx_ll_adc.h#L5216) in the ADC LL driver contains duplicated arguments in `&` chain:

```c
uint32_t AnalogWDMonitChannels = (READ_BIT(*preg, AWDy) & AWDy & ADC_AWD_CR_ALL_CHANNEL_MASK);
```

Since `READ_BIT` is expanded to that:

```c
((*preg) & AWDy))
```

The second `AWDy` usage is redundant. So all that expression can be simplified to that:

```c
uint32_t AnalogWDMonitChannels = (READ_BIT(*preg, AWDy) & ADC_AWD_CR_ALL_CHANNEL_MASK);
```

ASELSTM, 2021-02-19:
Hi @vpetrigo,

The issue you pointed out has been confirmed. Indeed, The second `AWDy `usage is redundant.
A fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2021-02-19:
ST Internal Reference: 101705

ASELSTM, 2023-04-06:
Hi @vpetrigo,

Thank you for your contribution. This issue has been fixed in the frame of version v1.17.1 of the STM32CubeL4. Please allow me then to close this thread.

With regards,

closed 2023-04-06 by ASELSTM