STMicroelectronics/STM32CubeL4 #32: MFX_INT_PIN pin is not defined

sfeutrier, 2021-02-18:
**Set-up**
 * Board : 32L4R9IDISCOVERY

**Issue**
When looking at the example BSP, in the main.c file there is this pin tested : MFX_INT_PIN. This pin is defined nowhere.
Moreover, when looking at the readme.txt, some files listed don't exist (and when looking at older commits, it seems they have never existed) like audio.bin and lcd_log_conf.h

RKOUSTM, 2021-02-23:
Hi @sfeutrier,

Thank you for  your contribution. Concerning the fist point, the `MFX_INT_PIN` interface pin is defined in the [stm32l4r9i_descovery.h](https://github.com/STMicroelectronics/STM32CubeL4/blob/master/Drivers/BSP/STM32L4R9I-Discovery/stm32l4r9i_discovery.h) file as follow:  
https://github.com/STMicroelectronics/STM32CubeL4/blob/d023c0d560ace11509f9b761c8913a9e48fcf194/Drivers/BSP/STM32L4R9I-Discovery/stm32l4r9i_discovery.h#L278

For the second point, the `audio.bin` file is available in the [AudioFile](https://github.com/STMicroelectronics/STM32CubeL4/tree/master/Projects/32L4R9IDISCOVERY/Examples/BSP/AudioFile) folder.  Your question about the missing of the lcd_log configuration template file (`lcd_log_conf.h`)  on the 32L4R9IDISCOVERY BSP example has been forwarded to our development team. We will get back to you as soon as we have more details.

Thank you again for your contribution. 

With regards,

sfeutrier, 2021-02-24:
Hi @RKOUSTM,

Okay I did not understand that somme files are in the Drivers folder and some in the Project folder. 
Maybe I missed it but it could be nice to put this information in the Project readme ?
Thank you for your support and for the git initiative.

Best regards

RKOUSTM, 2021-02-24:
Hi @sfeutrier ,

Thank you for your contribution. The issue you pointed out has been confirmed, a fix of the `lcd_log_conf.h `template file on the `32L4R9IDISCOVERY BSP example` will be implemented and made available in future release. 

Thank you once again for your contribution.

With regards,

RKOUSTM, 2021-02-24:
ST Internal Reference: 101981

RKOUSTM, 2021-12-15:
Hi @sfeutrier,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.17.1 of the STM32CubeL4 published recently on GitHub. Thank you again for having reported.

With regards,

closed 2021-12-15 by RKOUSTM