STMicroelectronics/STM32CubeL4 #33: How use CubeMX extract 32L4R9IDISCOVERY board demo?

zhouchuanfu, 2021-03-19:
Dear all:
    I want to use STM32CubeMx or STM32CubeIDE to generate official demo of the this board, but not find that at STM32CubeMX->New Project->Example Selector->Board.

ALABSTM, 2021-03-23:
Hi @zhouchuanfu,

Thank you for your question. Actually, the board name in Cube MX is `STM32L4R9I-DISCO`. I hope this helps.

You can report this to the Cube MX team, not through this support channel however. Indeed, this one is dedicated to aspects related to the software **published within** this repository. Aspects like those related to ecosystem and tools are rather treated at **ST Community** level, precisely within the [Cube MX](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) dedicated section in this case.

Please allow me to close this issue now. We are looking forward to reading from you.

With regards,

closed 2021-03-23 by ALABSTM

zhouchuanfu, 2021-03-23:
Hi @ALABSTM,
Thank you for your response, I will to go to **ST Community** post this question.
Thank you again.