STMicroelectronics/STM32CubeL4 #34: Legacy OPAMP_POWERMODE_NORMAL used in IS_OPAMP_POWERMODE()

CanastraRF, 2021-03-20:
I download STM32CubeL4 V1.17.0

In stm32l4xx_hal_opamp.h is in IS_OPAMP_POWERMODE() the legacy value OPAMP_POWERMODE_NORMAL used.

```
#define IS_OPAMP_POWERMODE(TRIMMING) (((TRIMMING) == OPAMP_POWERMODE_NORMAL) || \
                                      ((TRIMMING) == OPAMP_POWERMODE_LOWPOWER) )

```
But the current value are:

```
#define OPAMP_POWERMODE_NORMALPOWER   0x00000000U              /*!< OPAMP power mode normal */   
#define OPAMP_POWERMODE_LOWPOWER      OPAMP_CSR_OPALPM         /*!< OPAMP power mode low-power */

```
Please fix this ASAP
ReFe

ASELSTM, 2021-03-22:
Hi @CanastraRF,

The issue you pointed out is confirmed, an internal tracker has been logged and a fix will be implemented and made available in the frame of a future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2021-03-22:
ST Internal Reference: 103579

ASELSTM, 2023-04-06:
Hi @CanastraRF,

Thank you for your contribution. This issue has been fixed in the frame of version v1.17.1 of the STM32CubeL4. Please allow me then to close this thread.

With regards,

closed 2023-04-06 by ASELSTM