STMicroelectronics/STM32CubeL4 #37: stm32l4xx_hal_rcc embedding a directive within macro arguments is not portable

lsuchocki, 2021-06-30:
**Describe the set-up**
 * board: Nucleo32 L432KC
 * IDE: none / Makefiles
 * gcc version 9.3.1 20200408 (release) (GNU Arm Embedded Toolchain 9-2020-q2-update)
 * STM32CubeL4 v1.17.0, git commit 5e1553e07706491bd11f4edd304e093b6e4b83a4 (also present in v1.13.0)

**Describe the bug**
When compiling with "-Wpedantic", throws the following warning (error with -Werror):
```
Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_hal_rcc.c: In function 'HAL_RCC_OscConfig':
Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_hal_rcc.c:946:1: error: embedding a directive within macro arguments is not portable [-Werror]
  946 | #if defined(RCC_PLLP_SUPPORT)
      | ^
Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_hal_rcc.c:948:1: error: embedding a directive within macro arguments is not portable [-Werror]
  948 | #endif

```
**How To Reproduce**
Compile with additional compiler flags be adding "CFLAGS += -Wpedantic -Werror" to Makefile

**Additional context**
From the C standard, 6.10.3 Macro replacement:
11 [...] If there are sequences of preprocessing tokens within the list of arguments that would otherwise act as preprocessing directives, the behavior is undefined.

I believe the following patch could fix the issue (note line numbers are specific to v1.13.0):
```
--- a/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_hal_rcc.c       2021-06-30 17:51:13.772490643 -0500
+++ b/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_hal_rcc.c   2021-06-30 17:53:55.007765933 -0500
@@ -917,14 +917,20 @@
         }
 
         /* Configure the main PLL clock source, multiplication and division factors. */
+#if defined(RCC_PLLP_SUPPORT)
         __HAL_RCC_PLL_CONFIG(RCC_OscInitStruct->PLL.PLLSource,
                              RCC_OscInitStruct->PLL.PLLM,
                              RCC_OscInitStruct->PLL.PLLN,
-#if defined(RCC_PLLP_SUPPORT)
                              RCC_OscInitStruct->PLL.PLLP,
-#endif
                              RCC_OscInitStruct->PLL.PLLQ,
                              RCC_OscInitStruct->PLL.PLLR);
+#else
+       __HAL_RCC_PLL_CONFIG(RCC_OscInitStruct->PLL.PLLSource,
+                             RCC_OscInitStruct->PLL.PLLM,
+                             RCC_OscInitStruct->PLL.PLLN,
+                             RCC_OscInitStruct->PLL.PLLQ,
+                             RCC_OscInitStruct->PLL.PLLR);
+#endif
 
         /* Enable the main PLL. */
         __HAL_RCC_PLL_ENABLE();
 
```

RKOUSTM, 2021-07-02:
Hi @lsuchocki,

Thank you for your reports.  

According to the `Development toolchains and compilers` section of the release note (https://htmlpreview.github.io/?https://github.com/STMicroelectronics/STM32CubeL4/blob/master/Release_Notes.html), the STM32CubeIDE v1.5.1 - gcc version 7_2018_q2 and the STM32 System Workbench (SW4STM32) V2.7 toolchain are recommended for the STM32CubeL4 v1.17.0. 

Your report regarding the C standard requirements for the RCC driver, will be forwarded to our development team. We will get back to you as soon as we have more details.

Thank you again for your reports.

With regards,

RKOUSTM, 2021-07-05:
Hi @lsuchocki,

First we would like to thank you for your contribution. 

The issue you pointed out is confirmed. An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release to enhance the readability of the code. Also, this update is required by the gcc compiler. 

Thank you once again for your contribution.

With regards,

RKOUSTM, 2021-07-05:
ST Internal Reference: 110117

ALABSTM, 2021-12-15:
Hi @lsuchocki,

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.17.1** of the STM32CubeL4 published recently on GitHub. Thank you again for having reported.

With regards,

closed 2021-12-15 by ALABSTM