STMicroelectronics/STM32CubeL4 #38: USB continuous transfer IN-packets with length 0 bytes

AlexScient, 2021-07-03:
Hi STM team!

I found following bug. After connection my board to PC hid device registers correctly. When i call function USBD_CUSTOM_HID_SendReport(&hUsbDeviceFS, buf, 64) the data packet is received on PC and after that data packets with length 0 bytes are continuously transmitted to the PC.

System: 
Windows 10 Home
STM32CubeIDE v1.6.1
Repository:
STM32Cube_FW_L4_V1.17.0
Microcontroller:
STM32L432KBU6
Board: 
Custom with USB connector. 

I use USB CUSTOM HID Device with next settings:

in file usbd_conf.h
#define USBD_MAX_NUM_INTERFACES     1U
#define USBD_MAX_NUM_CONFIGURATION     1U
#define USBD_MAX_STR_DESC_SIZ     512U
#define USBD_DEBUG_LEVEL     0U
#define USBD_LPM_ENABLED     1U
#define USBD_SELF_POWERED     1U
#define USBD_CUSTOMHID_OUTREPORT_BUF_SIZE     64U
#define USBD_CUSTOM_HID_REPORT_DESC_SIZE     53U
#define CUSTOM_HID_FS_BINTERVAL     0x1U

in file usbd_custom_hid_if.c
__ALIGN_BEGIN static uint8_t CUSTOM_HID_ReportDesc_FS[USBD_CUSTOM_HID_REPORT_DESC_SIZE] __ALIGN_END =
{
  /* USER CODE BEGIN 0 */
		0x06, 0x00, 0xFF, 	// Usage page (Undefined)
		0x09, 0x01, 		// USAGE (Vendor usage 1)
		0xA1, 0x01, 		// COLLECTION (Application)
		0x85, 0x01,			// REPORT_ID (1)
		0x09, 0x01, 		// USAGE (Vendor usage 1)
		0x15, 0x00, 		// LOGICAL_MINIMUM (0)
		0x26, 0xFF, 0x00, 	// LOGICAL_MAXIMUM (255)
		0x75, 0x08, 		// REPORT_SIZE (8)
		0x95, 0x3F, 		// REPORT_COUNT (63)
		0x91, 0x82, 		// OUTPUT (Data, Var, Abs, Vol)
		0x85, 0x02,			// REPORT_ID (2)
		0x09, 0x02, 		// USAGE (Vendor usage 2)
		0x15, 0x00, 		// LOGICAL_MINIMUM (0)
		0x26, 0xFF, 0x00, 	// LOGICAL_MAXIMUM (255)
		0x75, 0x08, 		// REPORT_SIZE (8)
		0x95, 0x3F, 		// REPORT_COUNT (63)
		0x81, 0x82, 		// INPUT (Data, Var, Abs, Vol)
		0x85, 0x03,		// REPORT_ID (3)
		0x09, 0x03, 		// USAGE (Vendor usage 3)
		0x15, 0x00, 		// LOGICAL_MINIMUM (0)
		0x26, 0xFF, 0x00, 	// LOGICAL_MAXIMUM (255)
		0x75, 0x08, 		// REPORT_SIZE (8)
		0x95, 0x01, 		// REPORT_COUNT (1)
		0xB1, 0x82, 		// FEATURE (Data, Var, Abs, Vol)
  /* USER CODE END 0 */
  0xC0    /*     END_COLLECTION	             */
};

in file usbd_custom_hid_if.c
static int8_t CUSTOM_HID_OutEvent_FS(uint8_t event_idx, uint8_t state)
{
  /* USER CODE BEGIN 6 */
  UNUSED(event_idx);
  UNUSED(state);
  USBD_CUSTOM_HID_HandleTypeDef *hhid;
  if (hUsbDeviceFS.pClassData == NULL)
  {
	return (int8_t)USBD_FAIL;
  }
  hhid = (USBD_CUSTOM_HID_HandleTypeDef *)hUsbDeviceFS.pClassData;
  memcpy(&UsbRxBuff[0], hhid->Report_buf, USBD_CUSTOMHID_OUTREPORT_BUF_SIZE);
  /* Start next USB packet transfer once data processing is completed */
  USB_CustomHID_ReceiveCallback(&UsbRxBuff[0], USBD_CUSTOMHID_OUTREPORT_BUF_SIZE);
  USBD_CUSTOM_HID_ReceivePacket(&hUsbDeviceFS);
  return (USBD_OK);
  /* USER CODE END 6 */
}

in file usbd_customhid.c
#define CUSTOM_HID_EPIN_ADDR                         0x81U
#ifndef CUSTOM_HID_EPIN_SIZE
#define CUSTOM_HID_EPIN_SIZE                         64U
#endif
#define CUSTOM_HID_EPOUT_ADDR                        0x01U
#ifndef CUSTOM_HID_EPOUT_SIZE
#define CUSTOM_HID_EPOUT_SIZE                        64U
#endif
#define USB_CUSTOM_HID_CONFIG_DESC_SIZ               41U
#define USB_CUSTOM_HID_DESC_SIZ                      9U
#ifndef CUSTOM_HID_HS_BINTERVAL
#define CUSTOM_HID_HS_BINTERVAL                      0x05U
#endif /* CUSTOM_HID_HS_BINTERVAL */
#ifndef CUSTOM_HID_FS_BINTERVAL
#define CUSTOM_HID_FS_BINTERVAL                      0x01U
#endif /* CUSTOM_HID_FS_BINTERVAL */
#ifndef USBD_CUSTOMHID_OUTREPORT_BUF_SIZE
#define USBD_CUSTOMHID_OUTREPORT_BUF_SIZE            64U
#endif /* USBD_CUSTOMHID_OUTREPORT_BUF_SIZE */
#ifndef USBD_CUSTOM_HID_REPORT_DESC_SIZE
#define USBD_CUSTOM_HID_REPORT_DESC_SIZE             53U
#endif /* USBD_CUSTOM_HID_REPORT_DESC_SIZE */

MY SOLUTION: 
In the file stm32l4xx_hal.pcd.c i change next record
if ((ep->type == EP_TYPE_BULK) && ((wEPVal & USB_EP_KIND) == 0U))
to the
if ((ep->type != EP_TYPE_BULK) || ((ep->type == EP_TYPE_BULK) && ((wEPVal & USB_EP_KIND) == 0U)))
I think the transfer was continuously cause ep->type was not EP_TYPE_BULK and zero message length was not processed in any way.

ALABSTM, 2022-04-25:
Hi @AlexScient,

What you pointed out is a bug indeed. It has been already spotted and fixed internally. The fix should be made available in the frame of release **1.17.2** of this firmware.

With regards,

ALABSTM, 2022-04-29:
Hi @AlexScient,

The fix to this issue has been made available in the frame of release **1.17.2** of this firmware published a couple of days ago.

Please allow me to close this issue and thank you again for having reported.

With regards,

closed 2022-04-29 by ALABSTM