STMicroelectronics/STM32CubeL4 #44: HAL compiler warnings appear in Arduino IDE

zfields, 2021-08-24:
**Describe the bug**
Compiler warnings are generated when "Compiler warnings: All" is enabled.

```
In file included from /home/zak/.arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/libraries/SrcWrapper/src/LL/stm32yyxx_ll_adc.c:35:
/home/zak/.arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/system/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_ll_adc.c: In function 'LL_ADC_CommonDeInit':
/home/zak/.arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/system/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_ll_adc.c:310:53: warning: unused parameter 'ADCxy_COMMON' [-Wunused-parameter]
  310 | ErrorStatus LL_ADC_CommonDeInit(ADC_Common_TypeDef *ADCxy_COMMON)
      |                                 ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~

In file included from /home/zak/.arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/libraries/SrcWrapper/src/LL/stm32yyxx_ll_rng.c:23:
/home/zak/.arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/system/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_ll_rng.c: In function 'LL_RNG_DeInit':
/home/zak/.arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/system/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_ll_rng.c:87:40: warning: unused parameter 'RNGx' [-Wunused-parameter]
   87 | ErrorStatus LL_RNG_DeInit(RNG_TypeDef *RNGx)
      |                           ~~~~~~~~~~~~~^~~~
```

**To Reproduce**
Observed when using the standard `Servo.h` library.

Steps to reproduce the behavior:
1. Select an example sketch from the Servo library
2. Select "Compiler warnings: All" from the Arduino IDE preferences window.
3. Attempt to compile the selected example
4. See warnings

**Expected behavior**
No warnings should be generated from the board support library.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Pop_OS! 20.04
 - Arduino IDE version: v1.8.13
 - STM32 core version: v2.0.0
 - Upload method: N/A

**Board (please complete the following information):**
 - Name: Generic STM32L4 series
 - Hardware Revision: 
 - Extra hardware used if any: 

**Additional context**
This bug was first raised in `stm32duino/Arduino_Core_STM32`. It was explained as a bug in the HAL that flows through the `stm32duino` implementation, and the issue was requested to be reported here for an upstream fix.
https://github.com/stm32duino/Arduino_Core_STM32/issues/1484

ASELSTM, 2021-08-30:
Hi @zfields,

Thank you for this report. The issue you pointed out is confirmed, it will be tracked internally and a fix will be implemented and made available in a future release. To solve the issue you can apply the following fixes : 

```diff
ErrorStatus LL_ADC_CommonDeInit(ADC_Common_TypeDef *ADCxy_COMMON)
{
  /* Check the parameters */
  assert_param(IS_ADC_COMMON_INSTANCE(ADCxy_COMMON));

+ (void)(ADCxy_COMMON); 

  /* Force reset of ADC clock (core clock) */
  LL_AHB2_GRP1_ForceReset(LL_AHB2_GRP1_PERIPH_ADC);
```

```diff
ErrorStatus LL_RNG_DeInit(RNG_TypeDef *RNGx)
{
  /* Check the parameters */
  assert_param(IS_RNG_ALL_INSTANCE(RNGx));

+ (void)(RNGx); 

  /* Enable RNG reset state */
  LL_AHB2_GRP1_ForceReset(LL_AHB2_GRP1_PERIPH_RNG);
```

 Thank you once again for your contribution.

With regards,

ASELSTM, 2021-08-30:
ST Internal Reference: 112520

ASELSTM, 2021-08-30:
ST Internal Reference: 112522

ALABSTM, 2021-12-14:
Hi @zfields,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.17.1 of the STM32CubeL4 published recently on GitHub. Thank you again for having reported.

With regards,

closed 2021-12-14 by ALABSTM