STMicroelectronics/STM32CubeL4 #45: STM32L412 ADC Support broken (HAL not compiling)

gompf01, 2021-09-10:
**Short: STM32L412 does not have a configurable clock source at least in CubeMX 6.3 it is fixed to the SYSCLK and the HAL includes have it particularly disabled for STM32L412xx and STM32L422xx, but HAL still tries to configure it, which fails obviosuly** 

in **stm32l4xx_hal_rcc_ex.h** it is defined : 

`#if !defined(STM32L412xx) && !defined(STM32L422xx)`
`  uint32_t AdcClockSelection;      /*!< Specifies ADC interface clock source.`
`                                        This parameter can be a value of @ref RCCEx_ADC_Clock_Source */`

Yet the code in **stm32l4xx_hal_msp** at line 101 tries to configure it:

`    PeriphClkInit.PeriphClockSelection = RCC_PERIPHCLK_ADC;`
`    PeriphClkInit.AdcClockSelection = ADCCLockSelection;`
`    if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInit) != HAL_OK)`
`    {`
`      Error_Handler();`
`    }`

which breaks compiling:
`compiling stm32l4xx_hal_msp.c...`
`**stm32l4xx_hal_msp.c(101): error:  #136: struct "<unnamed>"  has no field "AdcClockSelection"**`
`      PeriphClkInit.AdcClockSelection = ADCCLockSelection;`
`stm32l4xx_hal_msp.c(101): error:  #20: identifier "ADCCLockSelection" is undefined`
`      PeriphClkInit.AdcClockSelection = ADCCLockSelection;`
`stm32l4xx_hal_msp.c: 0 warnings, 2 errors`

I am just switching to this IC, not sure if it was broken before, but it is broken in CubeMX6.3 with STM32Cube_FW_L4_V1.17.0

futurechin, 2021-09-13:
Yes, it is confirmed the CubeMX 6.3 with FW_L4_V1.17.0 on STM32L412 generate this error;
PeriphClkInit.AdcClockSelection = ADCCLockSelection;

It should be 
PeriphClkInit.AdcClockSelection = RCC_ADCCLKSOURCE_PLLSAI1;

gompf01, 2021-09-13:
> It should be
> PeriphClkInit.AdcClockSelection = RCC_ADCCLKSOURCE_PLLSAI1

Actually **AdcClockSelection** is **not defined** in the structure for L412 and L422 so i am not sure if this is the correct solution to add this field to the structure and set it as commented by Chin, or if the correct solution is totally different. 

/Gompf

futurechin, 2021-09-14:
You are right the AdcClockSelection is not defined for L412 and L422 because it is not structured in the PeriphClkInit. The PLLSAI1 is for L43x & L44x which I overlooked. 

If you refer stm32l4xx_hal_rcc_ex.h line 222 & 223 as below ; 
#if !defined(STM32L412xx) && !defined(STM32L422xx)
  uint32_t AdcClockSelection;      /*!< Specifies ADC interface clock source.

Below line should not be generated from CubeMX code generation when L412 or L422 is selected.
PeriphClkInit.AdcClockSelection = ADCCLockSelection;

gompf01, 2021-09-16:
Yes I think the entire clock section is not correct for the generated file. The question is if the Clock needs to be turned on elsewhere.

ASELSTM, 2021-09-21:
Hi @gompf01,

The point you reported is actually related to the CubeMX generation problem and not to the firmware published in this repository. Unfortunately, we don't treat aspect related to CubeMX tool in our GitHub repositories.  They are rather treated at the [STM32CubeMX](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) dedicated page of the **ST Community** . 

Please allow me then to close this thread. Thank you for your comprehension. 

With regards,

closed 2021-09-21 by ASELSTM