STMicroelectronics/STM32CubeL4 #50: B-L475E-IOT01A NFC Application

VVinter-melon, 2021-10-31:
Setup:
 * Board: B-L475E-IOT01A
 * STM32CubeIDE

Issue:
I am trying to configure the NFC module and ultimately trying to run the WriteTag demo.  I have imported the .h and .c libraries files from ../Common but faced several errors when building the project:
1.  In nfc_tt4_wrapper_template.h, line 31 `#include "component.h"`, component.h file does not exist anywhere, having also searched the STM32CubeL4 repo
2. nfc_tt4_wrapper_template.c creates a repeated definition of functions such as `uint16_t NFC_TT4_Initialization ( uint8_t* CCBuffer, uint8_t size )`. These functions have been defined in m24sr_wrapper.c

I would greatly appreciate any assistance with this matter. Thank you.

CIPop, 2021-12-16:
+1 I found the same issue with audio and other drivers related to this board: https://github.com/STMicroelectronics/STM32CubeL4/issues/58

ALABSTM, 2022-01-30:
Hi @VVinter-melon and @CIPop,

Thank you for this report. We will get back to you with a feedback as soon as possible. Please excuse the delay it may take us sometimes to reply. Thank you for your comprehension.

With regards,

RKOUSTM, 2022-02-25:
Hi @VVinter-melon,

Thank you for your report. An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release.

Thank you again for you report.

With regards,

RKOUSTM, 2022-02-28:
ST Internal Reference: 123566

ALABSTM, 2023-11-29:
Hi @VVinter-melon and @CIPop,

I hope you are doing well. Back to this question you asked. Actually, as **templates**, both [nfc_tt4_wrapper_template.h](https://github.com/STMicroelectronics/STM32CubeL4/blob/master/Projects/B-L475E-IOT01A/Applications/NFC/Common/NDEF_TagType4_lib/nfc_tt4_wrapper_template.h) and [nfc_tt4_wrapper_template.c](https://github.com/STMicroelectronics/STM32CubeL4/blob/master/Projects/B-L475E-IOT01A/Applications/NFC/Common/NDEF_TagType4_lib/nfc_tt4_wrapper_template.h) files are meant to be **customized** and **renamed** according to the application they are to be used into. Both [m24sr_wrapper.h](https://github.com/STMicroelectronics/STM32CubeL4/blob/master/Projects/B-L475E-IOT01A/Applications/NFC/Common/M24SR/m24sr_wrapper.h) and [m24sr_wrapper.c](https://github.com/STMicroelectronics/STM32CubeL4/blob/master/Projects/B-L475E-IOT01A/Applications/NFC/Common/M24SR/m24sr_wrapper.c) are examples of such a customization and renaming.

Mainly, the below line in `nfc_tt4_wrapper_template.h` that you mentioned above is not meant to be left "as-is"...
https://github.com/STMicroelectronics/STM32CubeL4/blob/f327fd77ebcf3ce295dcd47ba10450169a93856a/Projects/B-L475E-IOT01A/Applications/NFC/Common/NDEF_TagType4_lib/nfc_tt4_wrapper_template.h#L30

... and has been replaced by an actual file inclusion in `m24sr_wrapper.h` as you can see.
https://github.com/STMicroelectronics/STM32CubeL4/blob/f327fd77ebcf3ce295dcd47ba10450169a93856a/Projects/B-L475E-IOT01A/Applications/NFC/Common/M24SR/m24sr_wrapper.h#L32

This also explains why a function like `NFC_TT4_Initialization()` is defined both in `nfc_tt4_wrapper_template.c` and `m24sr_wrapper.c`. This is not repetition, rather redefinition.

The almost "empty" function body in `nfc_tt4_wrapper_template.c`...
https://github.com/STMicroelectronics/STM32CubeL4/blob/f327fd77ebcf3ce295dcd47ba10450169a93856a/Projects/B-L475E-IOT01A/Applications/NFC/Common/NDEF_TagType4_lib/nfc_tt4_wrapper_template.c#L97-L100

... has been replaced by an actual implementation in `m24sr_wrapper.c`.
https://github.com/STMicroelectronics/STM32CubeL4/blob/f327fd77ebcf3ce295dcd47ba10450169a93856a/Projects/B-L475E-IOT01A/Applications/NFC/Common/M24SR/m24sr_wrapper.c#L85-L140

I hope this answers your question. Please allow me to close this issue.

Thank you,

closed 2023-11-29 by ALABSTM

CIPop, 2023-11-29:
> I hope this answers your question. Please allow me to close this issue.

Thanks @ALABSTM for the detailed explanation! 

Would it be possible to at least add some comments in your template code explaining the same things as above?

E.g. this is the file Doxygen comment right now:

```
  * @file    nfc_tt4_wrapper_template.h
  * @author  MCD Application Team
  * @brief   This file makes the bridge between middleware generic NCF TagType4 
  *          function calls and the component.
```

While the filename "template" is a good indication, there isn't sufficient documentation here for someone to infer your explanation above.
Same goes for the `m24sr_wrapper.c` which could say that it is an implementation of the `nfc_tt4_wrapper_template.c` file.

ALABSTM, 2023-12-12:
Hi @CIPop,

I understand your request. I'll see what can bee done.

With regards,