STMicroelectronics/STM32CubeL4 #51: RCC_APB1ENR1_RTCAPBEN define missing from stm32l476xx.h

lucasholzen, 2021-11-03:
**Caution**
The current version of CMSIS (v5) is missing the define for the RTC APB1 clock enable.
https://raw.githubusercontent.com/STMicroelectronics/STM32CubeL4/master/Drivers/CMSIS/Device/ST/STM32L4xx/Include/stm32l486xx.h

It should look more like:

`
#define RCC_APB1ENR1_LCDEN_Pos               (9U)
#define RCC_APB1ENR1_LCDEN_Msk               (0x1UL << RCC_APB1ENR1_LCDEN_Pos) /*!< 0x00000200 */
#define RCC_APB1ENR1_LCDEN                   RCC_APB1ENR1_LCDEN_Msk
**#define RCC_APB1ENR1_RTCAPBEN_Pos            (10U)
#define RCC_APB1ENR1_RTCAPBEN_Msk            (0x1UL << RCC_APB1ENR1_RTCAPBEN_Pos) /*!< 0x00000400 */
#define RCC_APB1ENR1_RTCAPBEN                RCC_APB1ENR1_RTCAPBEN_Msk**
#define RCC_APB1ENR1_WWDGEN_Pos              (11U)
#define RCC_APB1ENR1_WWDGEN_Msk              (0x1UL << RCC_APB1ENR1_WWDGEN_Pos) /*!< 0x00000800 */
#define RCC_APB1ENR1_WWDGEN                  RCC_APB1ENR1_WWDGEN_Msk
`

Currently you have to manually define the registry location for the clock (or add the defines in yourself).

ALABSTM, 2022-04-19:
Hi @lucasholzen,

As you can see on the screenshot below from the [RM0351](https://www.st.com/resource/en/reference_manual/dm00083560-stm32l47xxx-stm32l48xxx-stm32l49xxx-and-stm32l4axxx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf), at page 255, _"this bit is **reserved** for STM32L4**7x**/L4**8x** devices"_. Hence, it is normal that it has been defined in none of the corresponding CMSIS files.

[image]

Would you mind reviewing the reason you need to define it? Probably there is no need.

With regards,

lucasholzen, 2022-04-19:
My apologies, I was having issues with RTC Alarm B at the time and they were unrelated to the CMSIS driver. I did not ending up having to define the bit at all.

ALABSTM, 2022-04-22:
Hi @lucasholzen,

If I understood correctly, you overcame the issue you were facing. Glad to read that.

With regards,