STMicroelectronics/STM32CubeL4 #52: IS_TIM_CCX_INSTANCE() doesn't compile when using LL TIM driver

benedekkupper, 2021-11-04:
**Describe the set-up**
 * STM32L432KC Nucleo
 * STM32CubeIDE 1.7.0
 * CubeL4 V1.17.0

**Describe the bug**
Using the LL TIM driver for all timers in the Cube project, when I try to use `IS_TIM_CCX_INSTANCE()` macro to check valid timer - channel number combination, the compilation fails:
` error: 'TIM_CHANNEL_1' was not declared in this scope`
(also for all other channel numbers)

**How To Reproduce**
1. Indicate the global behavior of your application project.
Compilation fails.
2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).
The problem is that `TIM_CHANNEL_n` are used in stm32l432xx.h and all the other CMSIS headers, but they are only defined in the TIM HAL driver header.
3. The use case that generates the problem.
Configure a timer, select LL driver, use  `IS_TIM_CCX_INSTANCE(TIM1, 1)` macro (the parameters are irrelevant).

**Additional context**
If you have a first analysis or patch correction, thank you to share your proposal.

ALABSTM, 2022-01-30:
Hi @benedekkupper,

Thank you for this report. We will get back to you with a feedback as soon as possible. Please excuse the delay it may take us sometimes to reply. Thank you for your comprehension.

With regards,

ASELSTM, 2022-02-24:
Hi @benedekkupper,

We are actually aware about the issue you raised out. It will be taken into consideration from our side. A fix will be available as part of coming series . Though, this may take time to be implemented on legacy series as the fix may have impacts on examples and applications provided within the STM32Cube package.  so thank you for your patience.  
As a workaround, you can include the HAL driver in you application to avoid compilation errors. 

Please allow me thus to close this thread. Thank you for you contribution and for you comprehension.

With regards,

closed 2022-02-24 by ASELSTM