STMicroelectronics/STM32CubeL4 #55: dsiPllInit.PLLNDIV too large for 32L4R9IDISCOVERY board

Gallium70, 2021-11-26:
**Describe the set-up**
 * The board: 32L4R9IDISCOVERY
 * IDE: PlatformIO + stm32cubel4 1.17.0 + gcc-arm-none-eabi 7.2.1

**Describe the bug**
The display on the screen seems compressed vertically, and bottom 2/3 of the display remains black.

[image]

**How To Reproduce**
1. Compile and upload [BSP Example](https://github.com/STMicroelectronics/STM32CubeL4/tree/master/Projects/32L4R9IDISCOVERY/Examples/BSP) ; The bug is also reproducible using [DSI Example](https://github.com/STMicroelectronics/STM32CubeL4/tree/master/Projects/32L4R9IDISCOVERY/Examples/DSI/DSI_CmdMode_SingleBuffer)

2. Change [PLLNDIV](https://github.com/STMicroelectronics/STM32CubeL4/blob/master/Drivers/BSP/STM32L4R9I-Discovery/stm32l4r9i_discovery_lcd.c#L295) to 100, then the display works fine. (For the DSI example, it's located [here](https://github.com/STMicroelectronics/STM32CubeL4/blob/master/Projects/32L4R9IDISCOVERY/Examples/DSI/DSI_CmdMode_SingleBuffer/Src/main.c#L310) )
[image]

3. If change PLLNDIV to 95, then about 1/5 bottom of the display remains black.

[image]

I suspect that the default 125 PLLNDIV is not able to generate a clock speed high enough for the display.

RKOUSTM, 2021-12-28:
Hi @Gallium70,

First we would like to thank you for your report.

We tried to reproduce the issue you described, but we couldn't notice the problem. 

Could you please confirm you are using the latest firmware v1.17.1 version and that the issue is systematically reproducible on your side?

Thank you again for your report.

With regards,

Gallium70, 2021-12-28:
Hi @RKOUSTM ,

Thanks for your reply.

At the time I posted this issue, the latest stable firmware version was v1.17.0. I have two l4r9i-disco boards and both of them could reproduce the issue then. The one in the picture stopped working several days ago though, so I can only try on the other board now.

I will try with v1.17.1 to see if the issue still exists.

Best regards,

RKOUSTM, 2022-01-06:
Hi @Gallium70,

 I hope you are fine. Any new feedback regarding this issue?

With regards,

Gallium70, 2022-01-07:
Hi @RKOUSTM  ,

I tried with a CubeMX generated Makefile project and programmed the board using CubeProgrammer, then everything works fine, with both v1.17.0 and v1.17.1. However, the issue still exists when using PlatformIO IDE, so maybe it's something related to PIO? But PIO should be using firmware code (v1.17.0) from this repo directly. That's weird.

Thanks for your help. I'll close this issue for now.

Best regards,

closed 2022-01-07 by Gallium70