STMicroelectronics/STM32CubeL4 #58: Incomplete B-L475E-IOT01 BSP/Drivers

CIPop, 2021-12-16:
**Describe the set-up**
 * B-L475E-IOT01 BSP
 * ARM GCC + CMake

**Describe the bug**
I am unable to find a complete repository with a compatible CMSIS+Drivers+BSP+HAL for B-L475E-IOT01.

B-L475E-IOT01 BSP/Drivers are outdated or missing in this repo. I found a few updated drivers here: https://github.com/STMicroelectronics/x-cube-iota1/tree/main/Drivers/BSP/B-L475E-IOT01 but that repo is unlikely the correct place as it is now changing to work with another board.

I've also opened a bug in the other repo: https://github.com/STMicroelectronics/x-cube-iota1/issues/2 

**How To Reproduce**
See https://github.com/STMicroelectronics/STM32CubeL4/compare/master...CIPop:merge_iot01 for a diff of what's currently missing.

CIPop, 2021-12-17:
Edit: 

@apalmieriGH clarified that this is the correct repo for the BSP. 

My understanding is that within this repo we only have the v1 BSP for this board which does not have audio driver support.
A preview v2 BSP with audio drivers is available in the x-cube-iota1 repo. The v2 BSP is incompatible with v1 and may lack some of the drivers. Merging the two BSPs is non-trivial.

An incomplete copy of the v2 BSP is available here: https://github.com/edgeimpulse/firmware-st-b-l475e-iot01a/tree/master/MP34DT01

RKOUSTM, 2022-01-18:
Hi @CIPop,

Thank you for your report. Unfortunately, there is no plan for our the next patch release STM32CubeL4 v1.17.2 to support the B-L475E-IOT01 BSP V2. I will keep you informed shall there be anything new.

Thank you for your report. We are looking forward to reading from you again.

With regards,

RKOUSTM, 2022-01-18:
Hi @CIPop,

Thank you for your report. Unfortunately, there is no plan for our the next patch release STM32CubeL4 v1.17.2 to support the B-L475E-IOT01 BSP V2. I will keep you informed shall there be anything new.

Please allow me to close this issue now. Thank you for your report. We are looking forward to reading from you again.

With regards,

closed 2022-01-18 by RKOUSTM