STMicroelectronics/STM32CubeL4 #61: 32L4R9IDISCOVERY FLASH_FastProgram; Problem with source=destination

gojimmypi, 2022-02-07:
**Describe the set-up**

I am using the VisualGDB extension for Visual Studio.

**Describe the bug**

I noticed the the `HAL_FLASH_Program` failed to program when the [source was the same as the destination](https://github.com/STMicroelectronics/STM32CubeL4/blob/f93a2f74f8e9912405dbf1a297b6df0c423eddf2/Projects/32L4R9IDISCOVERY/Examples/FLASH/FLASH_FastProgram/Src/main.c#L73). I have a [fix](https://github.com/STMicroelectronics/STM32CubeL4/pull/60) that puts the source data in a different variable, but someone may wish to look at the HAL and do some error checking if the source and destination addresses to program are then same.

**How To Reproduce**

Compile and run as-is.

**Screenshots**

Here, we see the `src_addr` address being set to same address of `Data64_To_Prog` in Flash. (which is not very interesting, particularly after the flash is [erased](https://github.com/STMicroelectronics/STM32CubeL4/blob/f93a2f74f8e9912405dbf1a297b6df0c423eddf2/Projects/32L4R9IDISCOVERY/Examples/FLASH/FLASH_FastProgram/Src/main.c#L111).

[image]

RKOUSTM, 2022-04-11:
ST Internal Reference: 83807

ALABSTM, 2023-04-20:
Hi @gojimmypi,

I hope you are doing well. Thank you for your proposal. Currently, there is no plan to consider this use-case.

Moreover, in the examples provided showing how to use the fast programming feature, it is indicated that the area to program should be in a different bank, as shown in the code snippet below.

https://github.com/STMicroelectronics/STM32CubeL4/blob/e691a513a48d0014f201ce35a265cbfca446396d/Projects/STM32L476G-EVAL/Examples/FLASH/FLASH_FastProgram/Src/main.c#L35-L37

Please allow me to close this issue. Thank you for your comprehension.

With regards,