STMicroelectronics/STM32CubeL4 #63: Missing License Text

gojimmypi, 2022-02-19:
There's a [note](https://github.com/STMicroelectronics/STM32CubeL4/blob/f93a2f74f8e9912405dbf1a297b6df0c423eddf2/Drivers/STM32L4xx_HAL_Driver/Inc/stm32l4xx_hal_flash.h#L12) that 

>  This software is licensed under terms that can be found in the LICENSE file in the root directory of this software component

However [that file](https://github.com/STMicroelectronics/STM32CubeL4/blob/master/Drivers/STM32L4xx_HAL_Driver/License.md) is now blank:

[image]

There's no license at [https://github.com/STMicroelectronics/STM32CubeL4/tree/master/Drivers](https://github.com/STMicroelectronics/STM32CubeL4/tree/master/Drivers)

Perhaps that text refers to the [repo license summary](https://github.com/STMicroelectronics/STM32CubeL4/blob/master/LICENSE.md) ? It seems a bit unusual to simply list the names of the licenses with no other text, but I am certainly not a license expert.

RKOUSTM, 2022-02-23:
Hi @gojimmypi,

Thank you for your contribution. You are absolutely right about this point. This point has already been raised internally and an update of the license text in the license.md file will be made available in the frame of a future release.

Thank you once more for your contribution.

With regards,

RKOUSTM, 2022-04-05:
Hi @gojimmypi,

Thank you for your contribution. The issue you pointed out is fixed and now available in GitHub. 

Please allow me then to close this thread. Thank you again for your contribution. We are looking forward to reading from you again.

With regards,

closed 2022-04-05 by RKOUSTM