STMicroelectronics/STM32CubeL4 #66: Error in LL_CRS_ConfigSynchronization function (L4 and G4)

Nikitarc, 2022-04-13:
**Describe the set-up**
STM32G4 LL driver V1.2.2 / 10-November-2021

**Describe the bug**
The function LL_CRS_ConfigSynchronization uses:
MODIFY_REG(CRS->CR, CRS_CR_TRIM, HSI48CalibrationValue) ;

So the calibration value is set in the lower byte of CRS->CR.
But CRS_CR_TRIM is in the high byte

So LL_CRS_ConfigSynchronization should use:
MODIFY_REG(CRS->CR, CRS_CR_TRIM, HSI48CalibrationValue << CRS_CR_TRIM_Pos) ;

The function LL_CRS_SetHSI48SmoothTrimming() uses the shift.

**How To Reproduce**
No need, read the code

**Additional context**
Missing << CRS_CR_TRIM_Pos

ASELSTM, 2022-04-19:
Hi @Nikitarc,

Thank for rising out this issue. Indeed, you are right and the issue is confirmed. The LL_CRS_ConfigSynchronization() should be updated as follow : 

```diff
__STATIC_INLINE void LL_CRS_ConfigSynchronization(uint32_t HSI48CalibrationValue, uint32_t ErrorLimitValue, uint32_t ReloadValue, uint32_t Settings)
{
-  MODIFY_REG(CRS->CR, CRS_CR_TRIM, HSI48CalibrationValue);
+  MODIFY_REG(CRS->CR, CRS_CR_TRIM, HSI48CalibrationValue << CRS_CR_TRIM_Pos);
  MODIFY_REG(CRS->CFGR,
             CRS_CFGR_RELOAD | CRS_CFGR_FELIM | CRS_CFGR_SYNCDIV | CRS_CFGR_SYNCSRC | CRS_CFGR_SYNCPOL,
             ReloadValue | (ErrorLimitValue << CRS_CFGR_FELIM_Pos) | Settings);
}
```

The fix will be implemented and made available in the frame of a future release.

With regards,

ASELSTM, 2022-04-19:
ST Internal Reference: 126601

Nikitarc, 2022-04-19:
Don't forget that this bug is in L4, G4, H7... at least

ASELSTM, 2023-04-07:
Hi @Nikitarc,

You can find the fix within this [commit](https://github.com/STMicroelectronics/stm32l4xx_hal_driver/commit/8d5c18d7ba3b6065dc103338b2ed176b6c74e0ad)  implemented on the [stm32l4xx_hal_driver](https://github.com/STMicroelectronics/stm32l4xx_hal_driver) repository as part of the rolling release process.  The fix will be available on the STM32CubeL4 repository in the frame of a future release.

With regards,

HBOSTM, 2023-07-21:
Hello @Nikitarc,

The fix you requested has been implemented and is now available in the frame of the latest package V1.18.0 release. This issue can be closed now. Thank you again for your contribution.

Best Regards,

closed 2023-07-21 by HBOSTM