STMicroelectronics/STM32CubeL4 #70: STM32L4xx_HAL_Driver/Inc/stm32l4xx_hal_gpio_ex.h: GPIO_AF14_TIM2 missing for STM32L4P5xx || STM32L4Q5xx

chrismas9, 2022-06-30:
Drivers/STM32L4xx_HAL_Driver/Inc/stm32l4xx_hal_gpio_ex.h
[Release v1.17.1](https://github.com/STMicroelectronics/STM32CubeL4/commit/f93a2f74f8e9912405dbf1a297b6df0c423eddf2)

At line 787 under `AF 14 selection` there is no entry for TIM2 for the STM32L4P5 and STM32L4Q5.
Every other L4 part in this file has `#define GPIO_AF14_TIM2         ((uint8_t)0x0E)  /* TIM2 Alternate Function mapping   */`
Accoding to the data sheet pin PA0, AF14 is TIM2_ETR. I have confirmed this in STM32CUBEMX by selecting this pin as external clock source. The sample code generated includes this in stm32l4xx_hal_msp.c, but there is no definition of GPIO_AF14_TIM2 in the HAL.

```
    /**TIM2 GPIO Configuration
    PA0     ------> TIM2_ETR
    */
    GPIO_InitStruct.Pin = GPIO_PIN_0;
    GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
    GPIO_InitStruct.Pull = GPIO_NOPULL;
    GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_LOW;
    GPIO_InitStruct.Alternate = GPIO_AF14_TIM2;
    HAL_GPIO_Init(GPIOA, &GPIO_InitStruct);

```
This problem was detected in MicroPython. I am currently porting all the STM32L4plus family to MicroPython and detected this because MicroPython enables all Timer pins and failed setting up PA0-TIM2_ETR.

L4P5.ioc attached demonstrates the problem. In the generated code `GPIO_AF14_TIM2` is used , but never defined.

[L4P5.zip](https://github.com/STMicroelectronics/STM32CubeL4/files/9021149/L4P5.zip)

ASELSTM, 2022-07-12:
ST Internal Reference: 131716

HBOSTM, 2023-07-13:
Hello @chrismas9 

The fix you requested has been implemented and is now available in the frame of the latest stm32l4_hal_driver package V1.18.0 release. This issue can be closed now. Thank you again for your contribution.

Best Regards,

closed 2023-07-13 by HBOSTM