STMicroelectronics/STM32CubeL4 #72: Shall the static flags be cleared as on the other control paths?

dima-kapustin, 2022-07-10:
https://github.com/STMicroelectronics/STM32CubeL4/blob/c5e83f31696c3da4fb374224471afd08d9d457b3/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_hal_mmc.c#L1728

There is the following statement on the other control paths in this function:

      /* Clear all the static flags */
      __HAL_MMC_CLEAR_FLAG(hmmc, SDMMC_STATIC_DATA_FLAGS);

However, it is not applied for DMA multi-block write mode for STM32L431 series...

Is it a bug or a feature?

Thanks.

ASELSTM, 2022-08-17:
ST Internal Reference: 133266

HBOSTM, 2023-07-21:
Hello @dima-kapustin,

The fix you requested has been implemented and is now available in the frame of the latest package V1.18.0 release. This issue can be closed now. Thank you again for your contribution.

Best Regards,

closed 2023-07-21 by HBOSTM