STMicroelectronics/STM32CubeL4 #75: HAL_OSPIM_Config incorrectly deactivates another instance

raitraak-rrk, 2022-09-08:
https://github.com/STMicroelectronics/STM32CubeL4/blob/c5e83f31696c3da4fb374224471afd08d9d457b3/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_hal_ospi.c#L2597

Configuration method incorrectly deactivates the other instance in case any of the following signals are turned off for both instances:
DQSPort = 0(disabled)
IOLowPort = HAL_OSPIM_IOPORT_NONE
IOHighPort = HAL_OSPIM_IOPORT_NONE

HBOSTM, 2022-11-09:
Hello @raitraak-rrk

Would you please give us more details about how you got this issue? Did you observed it at runtime ? if so, could you please share the code you have used to reproduce the problem.
With regards,

raitraak-rrk, 2022-11-11:
Hi,

Sharing the full code is not trivial. Pseudo code for same value of DQSPort:

OSPI_HandleTypeDef ospiHandle1;
OSPIM_CfgTypeDef ospimCfg1 { 1U, 0U, 1U, HAL_OSPIM_IOPORT_1_LOW,HAL_OSPIM_IOPORT_1_HIGH };
HAL_OSPIM_Config(&ospiHandle1, &ospimCfg1, HAL_OSPI_TIMEOUT_DEFAULT_VALUE);
OSPI_HandleTypeDef ospiHandle2;
OSPIM_CfgTypeDef ospimCfg2{ 2U, 0U, 2U, HAL_OSPIM_IOPORT_2_LOW,HAL_OSPIM_IOPORT_NONE };
HAL_OSPIM_Config(&ospiHandle2, &ospimCfg2, HAL_OSPI_TIMEOUT_DEFAULT_VALUE);

Second call to HAL_OSPIM_Config disables the first instance. Should be pretty obvious. Please let me know if you need more info.

KRASTM, 2024-02-01:
ST Internal Reference: 172216

KRASTM, 2024-03-29:
Fixed in d3b7dcfb72c6c6a046862654ef6e23cbed9935a8

closed 2024-03-29 by KRASTM