STMicroelectronics/STM32CubeL4 #78: Lack of __HAL_GPIO_EXT_DISABLE_IT or equivalent

ecrin4102, 2023-01-24:
In order to prevent a wake up from STOP mode from an exti line (connected to a GPIO for example), this is mandatory to mask the IRQ process on the EXTI stage. There is no other choice than using register directly with:
`EXTI->IMR1 &= ~(EXTI_LINE_0)`

Using only NVIC to disable IRQ will only prevent the execution of IrqHandler but will allow the wake up.
As all other peripherals has an __HAL_XXXXX_DISABLE_IT macro, maybe a __HAL_GPIO_EXT_DISABLE_IT macro can be created?

ecrin4102, 2023-01-24:
Refer to this thread for the lack :
[https://community.st.com/s/question/0D50X0000BdgWWISQ2/how-to-disablemask-an-exti-line-on-the-fly](url)
[https://community.st.com/s/question/0D50X00009XkfHdSAJ/disableenable-exti-interrupts-using-hal](url)

HBOSTM, 2023-02-06:
Hello @ecrin4102,

Firstly, thank you for this contribution. The EXTI enabling and disabling are handled at HAL_GPIO_Init() and HAL_GPIO_DeInit(). I hope you find in the explanations [here](https://github.com/STMicroelectronics/STM32CubeG4/issues/6) a satisfying answer to your question.

Best Regards,

HBOSTM, 2023-05-04:
Hi @ecrin4102,

Please allow me to close this issue. Thank you again for your comprehension.

With regards,