STMicroelectronics/STM32CubeL4 #80: TX IRQ remains active after HAL_I2C_Mem_Read_DMA

accabog, 2023-02-06:
Because HAL_I2C_Mem_Read_DMA MemAddress transmission is based on TX IRQ
https://github.com/STMicroelectronics/STM32CubeL4/blob/26be59071141d07a1d0dddb58b9fb335b027cbaa/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_hal_i2c.c#L3079

neither I2C_Mem_ISR_DMA nor I2C_ITMasterCplt will not disable the TX IRQ after the data transfer

https://github.com/STMicroelectronics/STM32CubeL4/blob/26be59071141d07a1d0dddb58b9fb335b027cbaa/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_hal_i2c.c#L5930

Therefore, if a non-IT based transmission such as HAL_I2C_Mem_Write will follow, a TX irq will be triggered without any interrupt callback defined, therefore the IRQ cannot be cleared and results in the program stuck in I2C ISR

accabog, 2023-02-12:
Hello @HBOSTM ,

To reproduce, try a HAL_I2C_Mem_Read_DMA and after that do HAL_I2C_Mem_Write.
The HAL_I2C_Mem_Write will trigger a TX IRQ, even though it should not.

HBOSTM, 2023-02-16:
ST Internal Reference: 145679

HBOSTM, 2023-07-13:
Hello @accabog 

The fix you requested has been implemented and is now available in the frame of the latest stm32l4_hal_driver package V1.18.0 release. This issue can be closed now. Thank you again for your contribution.

Best Regards,

closed 2023-07-13 by HBOSTM