STMicroelectronics/STM32CubeL4 #82: B-L475E-IOT01A1 WiFi

shijiameng, 2023-07-04:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * B-L475E-IOT01A1
 * STM32CubeIDE

**Describe the bug**
In [es_wifi_io.c](https://github.com/STMicroelectronics/STM32CubeL4/blob/master/Projects/B-L475E-IOT01A/Applications/WiFi/Common/Src/es_wifi_io.c), absence of bound check in line 213 function `SPI_WIFI_ResetModule` might cause stack buffer overflow:
https://github.com/STMicroelectronics/STM32CubeL4/blob/503ab076587f3241e5d8f25adea452dbf80a8b4f/Projects/B-L475E-IOT01A/Applications/WiFi/Common/Src/es_wifi_io.c#L211-L220

**How To Reproduce**

- Hardware preparation:  a wifi access point, set its SSID and passward as `LI_H3C` and `1qaz2wsx`

- Target application: `WiFi_Client_Server`

- Download [reproduce.zip](https://github.com/STMicroelectronics/STM32CubeL4/files/11945731/reproduce.zip), uncompress it and apply the patch. This patch includes the use case injection.

- Compile `WiFi_Client_Server` and run it with debugger. The program would halt at breakpoint `__BKPT(0x10)` if the buffer overflow arises.
 
**Additional context**
- Same issue in STM32H7 firmware package v1.11.0: https://github.com/STMicroelectronics/STM32CubeH7/issues/252
- A PR regarding this issue will be posted.

TOUNSTM, 2023-07-11:
Hello @shijiameng,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With regards,

ALABSTM, 2023-08-01:
ST Internal Reference: 158543