STMicroelectronics/STM32CubeL4 #85: How to find out the pinout and configuration or ioc. file in L496G-DISCROVERY DFSDM?

LinHuangYuan901013, 2023-09-25:
Hi !
I want to inquire the pinout and configuration setting detail in DFSDM L496G. Such as the the setting in channel, filter, clock. Because the situation is that I want to add DFSDM in another project, I can not just run the code in this example. 

When I try to get the information by the sample project, it seems that I can not find the .ioc file in the project when I open in CubeIDE; and even in CubeMX no ioc. file, I can not open the project without .ioc file.
Thanks for your reading

board: STM32L496G-DISCO

TOUNSTM, 2023-10-03:
Hello @LinHuangYuan901013,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With regards,

RJMSTM, 2023-12-22:
Hello,
Would you please share the whole project you have used to reproduce the issue in order to allow a better analysis of the problem.

With regards,

ALABSTM, 2024-05-27:
Hi @LinHuangYuan901013,

Thank you for your question. As indicated in the [CONTRIBUTING.md](https://github.com/STMicroelectronics/STM32CubeL4/blob/master/CONTRIBUTING.md#3-support-requests-and-questions), requests of support such as this one should be submitted to the [ST Community](https://community.st.com/t5/product-forums/ct-p/product-forums). There you should find the assistance you are looking and answers to your question.

In the meanwhile, as the `DFSDM Audio Record` is provided for different STM32L4 boards, a ticket will be logged at the attention of our development teams to ask for its integration into the `STM32CubeMX` tool.

With regards,

ALABSTM, 2024-05-27:
ST Internal Reference: 182421

ALABSTM, 2024-05-27:
Hi @LinHuangYuan901013,

A request has been logged internally. Hopefully, the `DFSDM Audio Record` should be available for `STM32L4` boards via the 'Example Selector' of a future version of the `STM32CubeMX` tool.

As there is no issue with the software provided within this repository, please allow me to close this thread. Thank you for your comprehension.

With regards,

closed 2024-05-27 by ALABSTM