STMicroelectronics/STM32CubeL4 #86: Use const parameters for initialization functions

PetteriAimonen, 2023-10-09:
The initialization functions such as [HAL_GPIO_Init](https://github.com/STMicroelectronics/STM32CubeL4/blob/master/Drivers/STM32L4xx_HAL_Driver/Inc/stm32l4xx_hal_gpio.h#L283) and [HAL_RCC_OscConfig](https://github.com/STMicroelectronics/STM32CubeL4/blob/master/Drivers/STM32L4xx_HAL_Driver/Inc/stm32l4xx_hal_rcc.h#L4836) do not modify the initialization structure given as argument.

The argument should be declared as `const`:

     HAL_StatusTypeDef HAL_RCC_OscConfig(const RCC_OscInitTypeDef *RCC_OscInitStruct);

Then user code can declare the struct `const` also, which will cause it to be placed in flash memory, saving RAM memory for other uses.

ALABSTM, 2023-10-13:
Duplicate of #15

ALABSTM, 2023-10-13:
Hi @PetteriAimonen,

Thank you for having reported. Actually, this topic has been already tackled in issue #15 and we are progressively deploying the use of the `const` qualifier in our drivers for all firmware. This takes some time to be done.

You can keep on tracking progress on this topic via the above-mentioned issue. Please allow me to close this one. Thank you for your comprehension.

With regards,

closed 2023-10-13 by ALABSTM

PetteriAimonen, 2023-10-13:
So that issue #15 is supposedly fixed in 2020, yet three years later this repository which has several releases since then still doesn't have the fix?