STMicroelectronics/STM32CubeL4 #89: Failed to return ES_WIFI_STATUS_OK of function ES_WIFI_HardResetModule

AReducer, 2024-06-05:
**Caution**
At the return point of ES_WIFI_HardResetModule  
[https://github.com/STMicroelectronics/STM32CubeL4/blob/692ff21c2b84943aa8e8859e71600b8dc23aaf92/Projects/B-L475E-IOT01A/Applications/WiFi/Common/Src/es_wifi.c#L1530](url)
```c
ES_WIFI_Status_t ES_WIFI_HardResetModule(ES_WIFIObject_t *Obj)
{
  int ret = 0;
  LOCK_WIFI();
  if (Obj->fops.IO_Init != NULL)
  {
    ret = Obj->fops.IO_Init(ES_WIFI_RESET); // SPI_WIFI_Init
  }
  UNLOCK_WIFI();
  return (ret > 0) ? ES_WIFI_STATUS_OK : ES_WIFI_STATUS_ERROR;
}
```

It requires the return value of `Obj->fops.IO_Init` (i. e. if it's set, `SPI_WIFI_Init`) as `ret >0`. However, when mode passed to `SPI_WIFI_Init` is `ES_WIFI_RESET`, it calls `SPI_WIFI_ResetModule` and return its return value.  The called reset module function only has two possible of return values, 0 or -1. So, the `ES_WIFI_HardResetModule` can only return the case of `ES_WIFI_STATUS_ERROR`.

**Describe the set-up**
 * stm32L475 discovery board
 * STM32 System Workbench

**Additional context**
change it to `ret == 0` may address this issue.

RJMSTM, 2024-06-10:
Hello @AReducer,

I've tested the application, and everything appears to be functioning correctly. Could you please provide more specific details about the problem you're encountering? Any additional information would be helpful to understand and address the issue more effectively.

Regards,

AReducer, 2024-06-10:
`Obj->fops.IO_Init` is called as
https://github.com/STMicroelectronics/STM32CubeL4/blob/692ff21c2b84943aa8e8859e71600b8dc23aaf92/Projects/B-L475E-IOT01A/Applications/WiFi/Common/Src/es_wifi_io.c#L144

then it return  SPI_WIFI_ResetModule 's return value `rc` https://github.com/STMicroelectronics/STM32CubeL4/blob/692ff21c2b84943aa8e8859e71600b8dc23aaf92/Projects/B-L475E-IOT01A/Applications/WiFi/Common/Src/es_wifi_io.c#L194

`rc` is `0` to be successful. 
https://github.com/STMicroelectronics/STM32CubeL4/blob/692ff21c2b84943aa8e8859e71600b8dc23aaf92/Projects/B-L475E-IOT01A/Applications/WiFi/Common/Src/es_wifi_io.c#L234

thus, `ret` here is 0 to be successful case. But it returns `ES_WIFI_STATUS_ERROR`. Shall it return `ES_WIFI_STATUS_OK`?
https://github.com/STMicroelectronics/STM32CubeL4/blob/692ff21c2b84943aa8e8859e71600b8dc23aaf92/Projects/B-L475E-IOT01A/Applications/WiFi/Common/Src/es_wifi.c#L1530

RJMSTM, 2024-07-05:
ST Internal Reference: 185903