STMicroelectronics/STM32CubeL4 #93: es-wifi: why not creating a submodule ?

joelguittet, 2024-08-05:
Hello,

The es-wifi driver is used twice here for the B-L475E-IOT01A and B-L4S5I-IOT01A and additionally in the X-CUBE-GNSS package at https://github.com/STMicroelectronics/x-cube-gnss1.

While it's not really a big library and so not a big issue, the most important is that they are all different!!! I created a PR on this repository to fix an UDP issue (see https://github.com/STMicroelectronics/STM32CubeL4/pull/92) but the X-CUBE-GNSS package as a similar one too.

Why not creating submodules when you have this kind of libraries ? Here for the es-wifi driver this is a perfect example where maintenance of this stuff will be greatly simplified.

Joel

ALABSTM, 2024-08-07:
Hi @joelguittet,

Thank you for this proposal. Actually, replacing `BSP` drivers by submodules is already in the pipe. However, the `es-wifi` looks to be a particular case as it is not located under `./Drivers/BSP/Components`.

I have first to check with our development teams. I will get back to you once I have their feedback.

With regards,

joelguittet, 2024-08-07:
Hello @ALABSTM 
Awesome 👍 thanks for this feedback.
Joel

ALABSTM, 2024-09-12:
ST Internal Reference: 190925

ALABSTM, 2024-09-24:
Hi @joelguittet,

As you may have noticed, the `es-wifi` driver is already provided within the [stm32-es-wifi](https://github.com/STMicroelectronics/stm32-es-wifi) repository.

However, the versions used within this firmware and the file tree structures are very different from the above-mentioned, making any attempt of alignment and use of submodule time and effort consuming, not to mention risks of side effects, which we cannot afford currently.

Please allow me to close this issue. Thank you for your comprehension and thank you again for your proposal.

With regards,

closed 2024-09-24 by ALABSTM

joelguittet, 2024-09-25:
Hello @ALABSTM 
Just checked very quickly the new repo, looks good to me. At least we can use es wifi source code without fetching the whole cube package, which is the initial intention !
So yes for me it's perfectly fine to close this issue now, thanks a lot for the work done in this topic !
Joel