STMicroelectronics/STM32CubeL4 #95: SDMMC SD_FindSCR() problem

by-gnome, 2024-10-29:
**Describe the set-up**
STM32L496VG, ICCARM v8.32, STM32Cube_FW_L4_V1.18.0.

**Describe the bug**
The problem was described and solved earlier in the STM32CubeF7, see [STM32CubeF7/issues/21](https://github.com/STMicroelectronics/STM32CubeF7/issues/21).
In STM32CubeL4 this problem is also present and can be solved in a similar way.

**Additional context**
Patch from STM32CubeF7 repository
```
--- a/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_hal_sd.c
+++ b/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_hal_sd.c
@@ -3962,13 +3962,17 @@ static uint32_t SD_FindSCR(SD_HandleTypeDef *hsd, uint32_t *pSCR)
     }
   }
 #else
-  while(!__HAL_SD_GET_FLAG(hsd, SDMMC_FLAG_RXOVERR | SDMMC_FLAG_DCRCFAIL | SDMMC_FLAG_DTIMEOUT | SDMMC_FLAG_DBCKEND))
+  while(!__HAL_SD_GET_FLAG(hsd, SDMMC_FLAG_RXOVERR | SDMMC_FLAG_DCRCFAIL | SDMMC_FLAG_DTIMEOUT))
   {
     if(__HAL_SD_GET_FLAG(hsd, SDMMC_FLAG_RXDAVL))
     {
       *(tempscr + index) = SDMMC_ReadFIFO(hsd->Instance);
       index++;
     }
+    else if(!__HAL_SD_GET_FLAG(hsd, SDMMC_FLAG_RXACT))
+    {
+      break;
+    }
 
     if((HAL_GetTick() - tickstart) >=  SDMMC_DATATIMEOUT)
     {
```

KRASTM, 2024-11-01:
Hello again @by-gnome,

Thank you for the report.

I will share your proposal to our team in order to analyze it, and to decide if it's the same case for F7. If you have additional details to share or if you encounter a problem, please feel free to do so. I will keep you inform if there are any updates.

With regards,

by-gnome, 2024-11-02:
This should be easy to reproduce:
- call BSP_SD_Init()
- make some reads and writes (or vice versa)
- call BSP_SD_DeInit()
- call BSP_SD_Init()
- you should stuck in SD_FindSCR()

KRASTM, 2024-11-06:
Hi @by-gnome,

I did some tests on Disco board even with the scenario you have mentioned, and I didn't encounter a problem any problem, especially in  SD_FindSCR(). Below an extract of the steps:

      SD_Initialize();
      FS_FileOperations();
      BSP_SD_DeInit();
      BSP_SD_Init();
     
However, I shared your proposal with the Development team, and I will keep you informed.

With regards,

KRASTM, 2024-11-06:
ST Internal Reference: 195806

by-gnome, 2024-11-06:
In my case after calling SD_FindSCR() the scr[1] contains the value 0 instead of 0x2358443, so SD_WideBus_Disable() returns HAL_SD_ERROR_REQUEST_NOT_APPLICABLE and HAL_SD_ConfigWideBusOperation(), HAL_SD_Init(), BSP_SD_Init() end with an error.