STMicroelectronics/STM32CubeL4 #97: warning: variable set but not used

CanastraRF, 2024-12-16:
Hi
My Compiler report

**../../HAL/STM32/STM32L4xx/stm32l4xx_hal_nand.c(525): warning: variable 'numpagesread' set but not used [-Wunused-but-set-variable]
  524 |   uint32_t numpagesread = 0U;
      |            ^
../../HAL/STM32/STM32L4xx/stm32l4xx_hal_nand.c(683): warning: variable 'numpagesread' set but not used [-Wunused-but-set-variable]
  682 |   uint32_t numpagesread = 0U;
      |            ^
../../HAL/STM32/STM32L4xx/stm32l4xx_hal_nand.c(851): warning: variable 'numpageswritten' set but not used [-Wunused-but-set-variable]
  850 |   uint32_t numpageswritten = 0U;
      |            ^
../../HAL/STM32/STM32L4xx/stm32l4xx_hal_nand.c(1004): warning: variable 'numpageswritten' set but not used [-Wunused-but-set-variable]
 1003 |   uint32_t numpageswritten = 0U;
      |            ^
../../HAL/STM32/STM32L4xx/stm32l4xx_hal_nand.c(1168): warning: variable 'numsparearearead' set but not used [-Wunused-but-set-variable]
 1167 |   uint32_t numsparearearead = 0U;
      |            ^
../../HAL/STM32/STM32L4xx/stm32l4xx_hal_nand.c(1333): warning: variable 'numsparearearead' set but not used [-Wunused-but-set-variable]
 1332 |   uint32_t numsparearearead = 0U;
      |            ^
../../HAL/STM32/STM32L4xx/stm32l4xx_hal_nand.c(1498): warning: variable 'numspareareawritten' set but not used [-Wunused-but-set-variable]
 1497 |   uint32_t numspareareawritten = 0U;
      |            ^
../../HAL/STM32/STM32L4xx/stm32l4xx_hal_nand.c(1661): warning: variable 'numspareareawritten' set but not used [-Wunused-but-set-variable]
 1660 |   uint32_t numspareareawritten = 0U;
      |            ^**

Please fix this?

KRASTM, 2024-12-20:
Hello @CanastraRF,

Thank you for the report.

Could you please, specify which compiler and version?

With regards,

CanastraRF, 2024-12-20:
Hi
We are migrating from ARMCC5 to ARMCLANG 6.22. Because ARMCC5 is no longer under support and some CMSIS Packages (RTOS2, RTX) are no longer available for ARMCC5.

KRASTM, 2025-01-06:
Hello @CanastraRF,

I checked with our team, and unfortunately, there are no plans to align the driver with AC6. Regarding the warnings, as mentioned in the release notes, you need to activate `AC5-like Warnings`, or alternatively, you can modify the code to resolve those warnings.

Please, allow me to close the issue, thank you for your comprehension.

With regards,

closed 2025-01-07 by KRASTM